Ultimaker/Cura #2690: Cura 3 imperfect prints

sidyoung, 2017-10-27:
My much of my is part printed on a 45 degree angle to avoid support structure. The portion that touches the bed is cylindrical in shape.  In the Beta 2.7 version it prints perfectly,  in version 3 it is sags roughly.
Both with the same settings.  I really like version 3 but I guess I am relegated to Beta 2.7

smartavionics, 2017-10-27:
Hi, could you please provide more information - it's not possible to understand your issue otherwise. If possible, please provide a screenshot of the model on the build platform and give an idea of what settings you are using. Thanks.

Thisismydigitalself, 2017-10-27:
I also installed and played with 3.0.x but never used it for printing. with all do respect and I have LOTs of respect, i feel 3.0.x is still half-backed. All my prints are still sliced with 2.7.

smartavionics, 2017-10-27:
> I also installed and played with 3.0.x but never used it for printing. with all do respect and I have LOTs of respect, i feel 3.0.x is still half-backed. All my prints are still sliced with 2.7.

Could you please provide an example where 3 performs worse than 2.7? Thanks.

Thisismydigitalself, 2017-10-28:
To start with, since i can't import 2.7 profiles and have 3.x show me only my exposed settings - that alone is a reason for me to not use it for now.  I hate wasting time in fixing stuff which should work right out of the box.

smartavionics, 2017-10-28:
> To start with, since i can't import 2.7 profiles and have 3.x show me only my exposed settings - that alone is a reason for me to not use it for now. I hate wasting time on fixing stuff which should work right out of the box.

Sorry, I can't help you then. I only know a little about the slicer back end, I know nothing about the UI, profiles, etc. Perhaps a UI/profile guru will be able to help?

Ghostkeeper, 2017-11-06:
Let's keep that discussion in https://github.com/Ultimaker/Cura/issues/2579, not here.

closed 2018-02-26 by ianpaschal