Ultimaker/Cura #2693: Cura 3.0.3 Does Not Slice

mbmast, 2017-10-27:
I load a small model in Cura 3.0.3.  In solid view, it renders fine.  On the information panel on the right it states "Ready to slice" but the model is not sliced.  When I change the view to Layer View, there are no layers. No matter where I position the layer slider, it reads that it is on layer 1.  

I've uploaded the small (450KB) STL file. 
[iPhone 6s Case.zip](https://github.com/Ultimaker/Cura/files/1423547/iPhone.6s.Case.zip)

Thisismydigitalself, 2017-10-28:
Just tested your model with my 3.0.3 and it was sliced as it should. no problems.

mbmast, 2017-10-28:
Right.  I’ve loaded models (STL files) and they render and slice and provide print time estimates and filament usage estimates.  But this one file, when I loaded it, Cura only rendered it as a solid.  It did not slice it, it did not show layers (it thought the model consisted of only 1 layer).  When I tweaked a profile setting (e.g. change the print speed from 70 to 71), then it sliced the model and displayed print times and filament usage.

 

There’s some glitch in there that occasionally prevents the slicing upon the initial load of an STL file.  The problem is intermittent.

fieldOfView, 2017-10-28:
I too have seen this happen a couple of times, but regarded it as a glitch in the master branch. Thanks for the report that this also happens in the released version.

sbts, 2017-10-29:
I get the same issue here. Some STL's don't slice on load, others do.
I generate my models with the current release of openscad. (2015 series)
Although I've noted the same from some thingiverse models as well

Ghostkeeper, 2017-11-06:
All of the developers and testers have seen this issue from time to time since at least the beginning of this year. But it occurs only twice per moon cycle and only when the sheep outside of our office building is eating semi-dry grass or something so we haven't found a way to debug this yet.

If you find a way in which we can reliably reproduce this with at least 20% probability or so, please tell us! This has been bugging me for a very long time.

ChrisTerBeke, 2017-11-06:
Oh Lord, the forsaken sheep...

[image]

souldrainer2021, 2017-11-27:
I found that this happens when supports with everywhere placement is activated on startup. When I disable and re-enable it, it is fixed.

Ghostkeeper, 2017-12-11:
No, that's not it. That setting is set to "Everywhere" by default but most of the time it slices fine. Indeed disabling and re-enabling it usually works.

ianpaschal, 2018-05-29:
JIRA ticket created, devs see: CURA-5419

diegopradogesto, 2019-09-10:
I gave another try to this issue and still is very hard to reproduce it. The internal ticket was also closed by lack of time and very hard to find the problem. 

I'm going to close this ticket, but if you still see this behaviour, please feel free to reopen it.

closed 2019-09-10 by diegopradogesto