Ultimaker/Cura #2695: cylinder with side thickness an even multiple of extrusion width leads to very poor slicing

dejaloomer, 2017-10-28:
I'm not sure which multiples apply but very easy to repro with a cylinder with a wall 3x the line width:
* I used fusion 360 to create a hollow cylinder with a shell thickness of 1.2
* Cura extrusion (line) width = .4
* Cura wall thickness = 1.2
I'd expect multiple layers of 3 smooth rings. Instead the slicing goes bonkers with the head moving all over the place and making tons of straight movement. The results of a print are pretty mangled. The first image shows the head movement. The second image is with line width set to .402 and results in a much more expected slice. Although the inner wall still does some unexpected straight movements.

[image]
[image]

smartavionics, 2017-10-28:
Try reducing the innner wall line width a little.

Thisismydigitalself, 2017-10-28:
Same here. modeling with F360, Printing with a 0.5 nozzle and i get similar results.

smartavionics, 2017-10-28:
Like this...
[image]

Thisismydigitalself, 2017-10-28:
should we "cheat" with thinner inner wall line width?..

Thisismydigitalself, 2017-10-28:
Ok, the 0.49 trick worked for me.
Just tried it and it works. why do i need to do that - that i don't know

[image]
[image]

DDDirk, 2017-10-28:
My guess is that the cause is that while we may see that it is supposed to be a circle, Cura just sees vertices and lines between them that 'cut corners', resulting in Cura thinking the inner wall won't fit, so it prints infill instead.
Most desirable would be the ability to feed splines to slicers and slicers putting splines in gcode. The resolution of, say, a G3 would then only be limited by the abilities of the printer (resolution of the motors).

Until then, some trickery is needed.

I also often see a conflict between wall line count and wall line width when I want something to be printed only as walls, for which I then fill in a wall count that is way too high (and no infill). Cura then often leaves gaps in the middle that I want filled.
For example, if I slice a cylinder with a wall 1.3 mm thick and in Cura all wall line widths 0.4 mm, I get only 3 walls. Interrestingly, though, if I make the wall 1.4 mm thick I do get 4 walls. So Cura doesn't allways blindly follow the wall line widths.

But it would be nice if Cura could be told to completely fill a model with walls.
Maybe by filling in an inner wall line width of 0. Right now, that gives an error (red field).

Or better, Cura nicely shows what affects what by hovering over a setting (a feature I really like). Right now, the wall line widths affect the wall count. Maybe there should be an option to reverse that. 
Which leaves the question what to do then if the wall line count is set too high to be realistic (eg 10 walls in the above example). Actually do that? Walls 0.13 mm thick? What if it is set at 50? Maybe give an error? Seems the simplest solution.

smartavionics, 2017-10-28:
I have created a PR that should help when walls are made from lines that completely fill the wall width. See https://github.com/Ultimaker/CuraEngine/pull/635.

dejaloomer, 2017-10-29:
Thanks for the quick reply and useful info. Setting inner wall to .39 does indeed result in a nice concentric slice. However the print head does still crisscross the tube in a interesting fashion. This results in a ton of strings across the tube. If I turn off combing the dark blue lines show light(see image). I guess this causes retraction so most of the strings go away but there are still some and a lot of burs and blobs on the inside where the head reversed. A tube with non-integral thickness (1mm) slices cleanly with no head traversal, so there does seem to be something pathological about the multiple case.
[image]

smartavionics, 2017-10-29:
If you are using Cura 3, enable the Optimize wall printing order setting that's in the experimental category. Also, to ensure that the z-seams of the red walls are aligned, select either shortest or user specified. Doing these things should ensure that the z-seams of all of the walls are aligned.

Ghostkeeper, 2017-11-06:
The mechanism that causes this is the "Fill Small Gaps" setting.

Since no more walls would fit in the space between these outer walls, it leaves them out. Now there's a gap that has to be filled using the Fill Small Gaps technique, which uses a smaller line width. However, the Fill Small Gaps algorithm has no notion any more of what lines or walls surround those gaps. Instead it just prints parallel lines similar to skin. It then tries to link together small line segments to create a passable result. This fails at two of the sides though since the parallel lines happen to be parallel to the walls as well, which makes those lines long so they can't be linked to the next lines any more.

With that in mind, I don't know a real solution to this that is safe in all cases and efficient to compute.

alekseisasin, 2017-11-27:
@dejaloomer , Do you have additional questions related to your issue, others we would like to close this issue.

Ghostkeeper, 2017-12-11:
The status of this issue is that it's basically the expected result but we know that it's less than optimal.

dejaloomer, 2018-01-13:
@alekseisasin No more questions. Thanks for the info. Is the protocol for me to close the issue because I'm fine with closing it.

fieldOfView, 2018-01-14:
Feel free to close it then.

closed 2018-01-15 by alekseisasin