Ultimaker/Cura #2698: Cannot open any project. Cura always imports objects only, no matter what.

filipgoc, 2017-10-28:
Cura 3.0.3
Mac 10.12.6 (Sierra)

- Select file in finder, open with: Cura, imports object automatically 
It worked once (since 3.0.3 installation) and since there never.

filipgoc, 2017-10-28:
Edit: I think this is only because I erroneously saved some files as .3mf and confused them with the .3mf that is a project. 

Trouble of course is it's the same on Mac now (and we all know why), so... that's the tradeoff.

filipgoc, 2017-10-28:
Edit2: Nope It was NOT the file confusion problem. 

New data: 

I managed to open several files as projects. All working as expected, loading both objects and settings. 

Then it stopped working again. Now I get the Project? dialog: 
[image]

But after that nothing happens. No project summary, no loading of settings, no loading of objects. Nothing.

filipgoc, 2017-10-28:
**Update 3:** Now it works again with the same exact files with which it did not work 10 minutes ago. Didn't even restart Cura. No clue.

ChrisTerBeke, 2017-10-31:
Closing until issue re-appears and can be debugged.

closed 2017-10-31 by ChrisTerBeke