Ultimaker/Cura #2699: Option to Default to a Named SD Card (Feature Suggestion)

sambartle, 2017-10-29:
It would be excellent if Cura had an option to default to a specific SD card (or cards) by name and always set those as the default "Save to Removable Device" destination when found.

For example.. Its common for users to have multiple removable devices connected all the time (I personally  always have between 9 and 11 removable devices plugged in) and Cura will simply pick the first alphabetically. 

The SD cards I use in my printer are all named "3DPrinter" so i can tell which one I am after. It would be great to be able to set an option in Cura where I can type that identifier in, and when Cura starts (or detects a new removable device inserted), it checks against the configured name and sets it as the new default if it matches the name entered.  

That way  a user could simply rename all the cards used in the printer and put one in, hit the button and be confident the file has saved to the correct card without me forgetting to switch the drive with the drop down arrow.

Ghostkeeper, 2017-11-06:
Or perhaps just remember the last-used one?

sambartle, 2017-11-07:
The only issue with remembering is that some cheap card readers (the type with multiple types of card support) tend to randomly reassign drive letters per type on each boot.. so it might not be the same next time. 

It would be a good start however as in most cases I guess it will usually be correct then.

Ghostkeeper, 2017-11-11:
We wouldn't remember the drive letter (as that is definitely Windows-specific) but rather the name of the drive.

fieldOfView, 2017-11-11:
I would think you would remember the path, and on Windows that path is the driveletter.

Ghostkeeper, 2017-11-17:
Well that's a possibility, but if it gets allocated to a different drive letter you'll save it to the wrong drive then. But we can request the drive's metadata, maybe? Get a name from there?

fieldOfView, 2017-11-18:
On Windows, driveletters should not change. A bigger problem would be people using multiple sd cards (eg for multiple printers) with different labels.

Anyhow, on Windows it is customary to store drive letters in recent paths, not volume labels.

Ghostkeeper, 2018-02-05:
Our project manager just removed this from our planning because it's older than 12 weeks. Or rather, it will be older than 12 weeks next week... (?)

sambartle, 2018-02-05:
That doesn't seem like a reason to remove it.. but I guess its low priority so other stuff needs doing first. 

Ill have a look at adding the feature myself when I have time as its a constant frustration.

Ghostkeeper, 2020-08-10:
This was implemented in Cura 4.7. It should now remember the output device it was saved to last.

closed 2020-08-10 by Ghostkeeper