Ultimaker/Cura #2701: In Cura 3.x Mac, how do I export a JSON settings file

edasque, 2017-10-29:
I am trying to use the CLI to slice models but it's asking me for a JSON settings files and all the files I find are of CFG type (headers, properties).

```
/Applications/Cura.app/Contents/MacOS/CuraEngine slice -j settings.json -l output_stl/output_${i}.stl -o output_stl/output_${i}.gcode
```

nallath, 2017-10-30:
You need to use the definition file.

edasque, 2017-10-30:
@nallath, can you elaborate? How do you obtain that tile from the printer selection & custom print settings I have set up in Cura GUI?

nallath, 2017-10-30:
You can't get it from the GUI. The command line interface is just something that we use for debugging. The .json file that it needs is the fdmprinter.json file, which describes what the defaults of the settings are. You can set the value of a specific setting by command line by using -s (you can get a full list of commands by just running the curaEngine without any commands)

edasque, 2017-10-30:
So there isn't really a way for me to take my current UI settings to slice something in batch through the command line?

nallath, 2017-10-30:
If you look in the logs, you can see that it prints a list of settings formatted in such a way so that it can be passed to the command line engine.

edasque, 2017-10-30:
Such as these:
```
-s machine_head_with_fans_polygon="[[-20, 10], [10, 10], [10, -10], [-20, -10]]" -s z_seam_corner="z_seam_corner_inner" -s raft_interface_jerk="20" -s skin_overlap_mm="0.02"
```

Should I be using CuraEngine for batching instead of Cura?

nallath, 2017-10-30:
Cura (the front-end that is) does not have the option to batch slice g-codes. This could be created with a plugin (i've done it for an internal project once), but that requires a bit of programming.
So if you want as little coding as possible, the engine is indeed the way to go, even though you are using a debug interface for it.

edasque, 2017-10-31:
I might go the plugin way, might end up being easiest if not as easy to integrate into a Makefile (doing OpenScad export of *n* stl first).

Before I do, is there a way to get all those -s key=value easily into that json file?

Also, I chained -s from the log, adding ```-j /Applications/Cura.app/Contents/Resources/resources/definitions/fdmprinter.def.json \```

And I get a gcode produced. I can open it Cura and it looks 'good'. But the model isn't centered (and flat on the base, which it would be if I opened the STL directly in the UI). Am I missing something?

nallath, 2017-10-31:
The CuraEngine doesn't do centering of models. The front-end does some tricks to ensure that the model is sitting on top of the build plate. 

As for the values, you can just change the default_value properties in the fdmprinter.def.json file. Those are the values that it uses.

Ghostkeeper, 2017-11-07:
You can use `mesh_position_x`, `mesh_position_y` and `mesh_position_z` to reposition the model from the command line. These get added to the X, Y and Z coordinates of the vertices of the model.

edasque, 2017-11-08:
Thank you, @Ghostkeeper , I'll try that.

closed 2017-11-08 by edasque