Ultimaker/Cura #2702: Top/Bottom interface not graying out

ChrisRiddell, 2017-10-30:
Cura 3.0.3
Windows 10

When you change the Top or Bottom Layers it does not gray out Top/Bottom Thickness or the Top Thickness and or Bottom Thickness, Like it does for Wall Thickness.

I think it should have the same behavior as Wall Thickness or it should update the mm value to match what is inputted into Top layers, Bottom Layers.

Ghostkeeper, 2017-11-06:
Yup, this reproduces. The disabling of the setting seems to propagate only one generation.

Ghostkeeper, 2018-02-05:
Our project manager removed this bug from our planning because it's older than 12 weeks. Seems it'll stay broken for a while.

ChrisRiddell, 2018-02-05:
I'm starting to get the impression Ultimaker does not really care about fixing bugs that are not seen a show stoppers and only really want to focus on new and experimental features.

What makes great software good is all the little things as they do start adding up over time.

closed 2023-06-24 by github-actions[bot]