Ultimaker/Cura #2704: interesting combing fail. goes to origin many times for no apparent reasons

filipgoc, 2017-10-31:
combing obviously completely messes things up by going to the origin many times for some reason. curious about ideas.  
- this happens no matter where on the build plate the object is positioned. 

Combing is off
[image]

Combing is on
[image]

Combing is off for skin.
[image]

filipgoc, 2017-10-31:
A project file for those interested: 

[combing_origin.3mf.zip](https://github.com/Ultimaker/Cura/files/1429281/combing_origin.3mf.zip)

EliW, 2017-10-31:
I have seen the same behavior on my Rostock Max v3 using combing (just in my case it's going to 'center' 0,0 of the circle.  But exact same thing.  I was trying to figure out why it was doing that myself.   It also will regularly seem to just at travel speed run the entire border of the piece.

smartavionics, 2017-10-31:
I looked at your project file and this is what I found... I don't get the travels to the origin that you do but I do get a bunch of error messages that look like this:
````
2017-10-31 07:48:08,561 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] [ERROR] Clipper::offset failed. See generated debug.html!
2017-10-31 07:48:08,561 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Black is original
2017-10-31 07:48:08,561 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Blue is offsetted polygon
````
They come from a method called PolygonUtils::ensureInsideOrOutside(). Interestingly, that function behaves differently when DEBUG is enabled and I don't just mean that it prints out the messages, it also returns a different value. So perhaps it should always return the value it does when DEBUG is defined when it detects the situation it is whining about.

BTW, completely unrelated to this issue - you can print that model without support if you add a thin solid layer at the point where the support ends. I do that all the time for holes in horizontal surfaces that have tubes above.

filipgoc, 2017-11-06:
interesting things. 

On the unrelated issue @smartavionics , definitely a good trick so thanks for mentioning it!

I also often add a 'fake roof' when elevated holes are present, but in this specific case I prefer a bit of support as it's still arguably less post processing (less tools) and I don't care about the surface being nice here.

I used to print this with simplify3d's support, and it worked so well I just never felt the need to add the roof. Now to replicate it in Cura...

smartavionics, 2017-11-06:
> I also often add a 'fake roof' when elevated holes are present, but in this specific case I prefer a bit of support as it's still arguably less post processing (less tools) and I don't care about the surface being nice here.

Understood.

As to your original problem, I can't actually reproduce the problem here at the moment. Sorry, not much use, I know.

smartavionics, 2017-11-06:
I have created a PR that works around this issue (https://github.com/Ultimaker/CuraEngine/pull/645). I hav en't bothered to really understand what's going wrong here but that fix does zap the weird travels back to 0,0.

smartavionics, 2017-11-06:
FYI - the critical setting value that triggers this problem is the outer wall line width. Reducing it to 0.47 (from 0.49 as set in your profile) makes the wacky moves go away.

smartavionics, 2017-11-07:
The PR I mentioned above has been merged into the backend and so the next release of Cura should no longer behave like this. Thanks a lot for taking the time to report the bug and provide decent info. Most people just whine about something not being right without bothering to actually provide enough info to reproduce the problem (and then they wonder why the bug/feature doesn't get fixed!)

Ghostkeeper, 2017-11-26:
Loading that project file into the 3.1 beta causes no more weird lines to zero for me.

closed 2018-03-02 by ChrisTerBeke