Ultimaker/Cura #2705: Always fails when slicing with a slot in part

PukinDog, 2017-10-31:
I have done several things that have slots in them. All is done in Solidworks. When the slot is oriented as shown in the pictures, Cura always fail at the start of the slot. It's not just this part it's in several parts. If I try in another slicer it prints just fine so the part or the printer is eliminated as the source of the problem.
[image]
[image]
[image]

smartavionics, 2017-10-31:
Hi, could you please provide the above model so I can see where the problem lies.

PukinDog, 2017-10-31:
Yes. As stl in the zip. If you need any other format just tell me!
Thanks
Regards
Johan
[BLTouch Mounting V2.zip](https://github.com/Ultimaker/Cura/files/1430839/BLTouch.Mounting.V2.zip)

smartavionics, 2017-10-31:
Hi, given the overhang, I wouldn't expect this to print well without support enabled. You are using support?

[image]

PukinDog, 2017-10-31:
No I don't use support. And several other objects that have slots and are printed in this orientation gets exactly the same error. It's really strange. Neither they have used support when printing. I have tried a wide sort of settings but to no avail.

smartavionics, 2017-10-31:
It's not the slot that's the problem, it's the little "beak" at the front of the slot. If you didn't have that feature, I would expect that model to print fine with no support required.

PukinDog, 2017-10-31:
I'm sorry to say so but other parts haven't that beak and they fail to in just the same way.
I removed the beak for test.
[No beak.zip](https://github.com/Ultimaker/Cura/files/1431015/No.beak.zip)

smartavionics, 2017-10-31:
Thanks for the no-beak model - I printed one using my default settings and it came out perfectly usable

[image]
[image]

The settings used were:

````
;SETTING_3 {"global_quality": "[general]\\nversion = 2\\nname = Kossel Mini (hot
;SETTING_3  bed)\\ndefinition = kossel_mini_mb\\n\\n[metadata]\\nquality_type = 
;SETTING_3 normal\\ntype = quality_changes\\nsetting_version = 4\\n\\n[values]\\
;SETTING_3 nacceleration_enabled = False\\nacceleration_wall = None\\nadhesion_t
;SETTING_3 ype = skirt\\ncool_fan_full_at_height = 3\\ncool_fan_min_at_height = 
;SETTING_3 1\\ncool_fan_speed = 80.0\\ncool_min_layer_time = 15\\ncool_min_layer
;SETTING_3 _time_fan_speed_max = 15\\ncool_min_speed = 15\\nexpand_skins_expand_
;SETTING_3 distance = 2\\nfill_perimeter_gaps = nowhere\\ninfill_before_walls = 
;SETTING_3 False\\ninfill_line_width = 0.8\\ninfill_overlap = 10\\ninfill_patter
;SETTING_3 n = zigzag\\ninfill_sparse_density = 30\\njerk_enabled = False\\njerk
;SETTING_3 _wall = 5\\nlayer_height = 0.2\\nlayer_height_0 = 0.25\\nline_width =
;SETTING_3  0.5\\nmaterial_bed_temperature = 45\\nmaterial_diameter = 1.75\\nmat
;SETTING_3 erial_print_temperature = 190\\nmaterial_print_temperature_layer_0 = 
;SETTING_3 200\\nmin_skin_width_for_expansion = 0.1\\noptimize_wall_printing_ord
;SETTING_3 er = True\\nretraction_amount = 6\\nretraction_combing = noskin\\nret
;SETTING_3 raction_extra_prime_amount = 0\\nretraction_hop = 0.5\\nretraction_ho
;SETTING_3 p_enabled = True\\nretraction_min_travel = 2.5\\nretraction_prime_spe
;SETTING_3 ed = =retraction_speed\\nretraction_retract_speed = =retraction_speed
;SETTING_3 \\nretraction_speed = 30\\nskin_outline_count = 0\\nskin_overlap = No
;SETTING_3 ne\\nskirt_brim_minimal_length = 500\\nskirt_gap = 8\\nspeed_infill =
;SETTING_3  60\\nspeed_layer_0 = =speed_print * 30 / 60\\nspeed_print = 60\\nspe
;SETTING_3 ed_slowdown_layers = 1\\nspeed_topbottom = 40.0\\nspeed_travel = 300\
;SETTING_3 \nspeed_wall_0 = 40.0\\nspeed_wall_x = 60.0\\nspeed_walls_layer_0 = 2
;SETTING_3 0.0\\nsupport_angle = 60\\nsupport_enable = False\\nsupport_infill_ra
;SETTING_3 te = 25\\nsupport_interface_enable = True\\nsupport_interface_height 
;SETTING_3 = 0.6\\nsupport_pattern = zigzag\\nsupport_type = buildplate\\nsuppor
;SETTING_3 t_xy_distance = 1\\nsupport_z_distance = 0.2\\ntop_bottom_pattern = l
;SETTING_3 ines\\ntop_bottom_thickness = 0.6\\ntravel_compensate_overlapping_wal
;SETTING_3 ls_enabled = True\\nwall_0_wipe_dist = 0\\nwall_thickness = 1\\nz_sea
;SETTING_3 m_corner = z_seam_corner_inner\\nz_seam_relative = True\\nz_seam_type
;SETTING_3  = sharpest_corner\\nz_seam_x = 0\\nz_seam_y = 50\\n\\n"}

````

smartavionics, 2017-10-31:
I should say that there's a few settings in there that the standard cura doesn't support so if you don't recognise something, just ignore it!

And I didn't use support.

PukinDog, 2017-10-31:
This is strange. I haven't got any of the objects to print. I will test this tomorrow and report back how it went. Is there something in the Marlin firmware that can mess this up?

smartavionics, 2017-10-31:
Sorry, I don't know anything about Marlin.

Could you please slice the same model and upload the gcode sometime, perhaps I can spot what's wrong by looking at that?

closed 2017-10-31 by ChrisTerBeke

PukinDog, 2017-10-31:
Yes. I can do that later tonight.
And I thought about it. There is no need for me to printing the part with
no slot since that is not how the part looks. And other slicer can printing
it with no problem. That makes my thought about Marlin problem not valid
either.
I'll send you the gcode and hope that you can help me find a solution since
I like the program.

PukinDog, 2017-10-31:
Why is this is sur closed? A solution is not found.

ChrisTerBeke, 2017-10-31:
As it's not an issue in Cura and we'd like to keep the issue list clean (can re-open if that's the case but I doubt it). You can still use this thread to continue your conversation.

PukinDog, 2017-10-31:
I think it is since it is only with Cura it fails. Hard to think it's
something else  then in my mind.

smartavionics, 2017-10-31:
Sure, let's keep an open mind but at the moment it looks like a settings problem. Please post the gcode for that part when you can and maybe we can work out where it's going wrong. At the moment, we don't need to re-open the issue to keep talking about it.

PukinDog, 2017-11-01:
[BLTouch Mounting V2.zip](https://github.com/Ultimaker/Cura/files/1432784/BLTouch.Mounting.V2.zip)
This is the Gcode that fails. Nothing special with it. Note that the exact same model sliced with another program prints ok with the same settings.
/Johan

ChrisTerBeke, 2017-11-01:
A bit clarification: we don't see 'wrong' settings as a Cura issue but more of a support thing. If it's indeed a bug in the Cura source code we'll re-open.

smartavionics, 2017-11-01:
Hi, thanks for the gcode.

Here are images of the gcodes showing the print speed as colour:

[image]
[image]

You can see that my gcode prints everything above the base slower.

The main differences in that from what I am doing are:

1 - layer height is smaller (probably not critical)
2 - infill printed before walls (could make a difference)
3 - infill density a little higher (could make a difference)
4 - cooling fan running faster (could make a difference)
5 - no minimum layer time (I use 15 seconds - try using this)

I think that your problem is that the vertical parts above the base are being printed too quickly and maybe you are using too much cooling fan.
6 - no minimum print speed (I use 15 mm/s - try using this)

PukinDog, 2017-11-02:
Of course that wrong settings isn't a bug in the program, that would be strange. :)
I do use the same settings in Simplify3D though.

How do I set #5 and #6 ? I can't find those settings?

Regards
Johan

smartavionics, 2017-11-02:
> How do I set #5 and #6 ? I can't find those settings?

They are on the cooling tab and you can also find the fan settings there so you can experiment with using less/more fan.

PukinDog, 2017-11-04:
I will try this. I guess that it might work.
Or to make it simple. Did you manage to print the first file I sent you with the "beak" on it with these settings? And with no artifacts or bad layer bonding?
If you did then it is truly these settings I have to change. Or slice everything with a slot in Simplify3D so I don't have to think about it.
Regards
Johan

smartavionics, 2017-11-04:
I haven't tried printing the beaked version because unless I give it support the beak won't print well. But I would not expect the quality of the rest of the print to change when support is added. When I printed the non beak version there is perfect layer bonding and very little in the way of artifacts.

Ghostkeeper, 2017-11-10:
Seeing as how it did apparently work in other slicers, I'm guessing that this eventually boils down to Cura's limited bridging capability.

PukinDog, 2017-11-10:
I guess you're right.

smartavionics, 2017-11-10:
I agree that Cura's bridging ability is limited but that's not the main problem here. Look at the pictures at the top of this thread and you can see that there is delamination at the bottom of the slot, that's nothing to do with bridging performance. I have printed this piece (sans-beak) and it came out pretty much perfect. Certainly with no delamination or dodgy layers at the bottom of the slot like we see above. So I made some suggestions as to how the settings could be altered by the OP to achieve better results. Has that been tried? Did it make any difference?

PukinDog, 2017-11-11:
I haven't had time yet since I had several other things to print and used Simplify3D instead.
I noticed that the settings you suggest are present in S3D also but I don't have them switched on there. I also print them without support and the beak prints without problem.
I don't know at this point what the problem is but from tests with the same settings Cura has some problem. I don't have the first file where I first noticed this behavior on since I redesigned it. But that didn't had a beak but a slot and the whole layer just beneath the slot de-laminated completely. Even that part S3D could print with the same settings.
Perhaps one has to tinkle a little more with the settings in Cura to get it to work? It would be really interesting to see your print on the actual part with the beak and not with support and using the settings you suggests! I mean, you are a pro on the software so the settings you use to make it work is interesting.
/Johan

smartavionics, 2017-11-11:
OK, I printed the model with the beak. I didn't use any support so the beak at the top looks poor but I expected that due to the overhang.

[image]

[image]

There's absolutely no delamination. The part is strong. I really don't think that Cura has a problem slicing this part.

smartavionics, 2017-11-11:
And for completeness, I printed it again using support. The beak is better, otherwise no difference from the non-support version.

[image]