Ultimaker/Cura #2707: "Save to removable drive" dropdown on Linux lists non-removable drives

lokster, 2017-11-01:
There is not much to explain here - the title says it all.
If you use Cura on Linux, the "Save to removable drive" drop down lists all disk drives, no matter internal or external.
The issue is made worse by the fact that it does not allow you to define some "preferred" device to be auto selected if present. This is in relation to #2699

nallath, 2017-11-01:
It doesn't do this for all linux versions. Im on Fedora, but it doesn't occur there.

awhiemstra, 2017-11-01:
The code for the removable drive detection is really naive and simple, it basically just lists the directories /media and /run/{user}/media. This will indeed include NFS shares, fixed disks and similar things mounted under those locations. Ideally, we would be using UDisks for this instead, but we never really got around to that. But UDisks at least has the capability to discern between removable disks and other things.

Ghostkeeper, 2017-11-06:
The relevant code is here if someone wants to fiddle with it:
https://github.com/Ultimaker/Cura/blob/c61b9c41ab03e7b0a723cd1f70a4cfd5bbf88b81/plugins/RemovableDriveOutputDevice/LinuxRemovableDrivePlugin.py#L21-L31

And here's an example of how to enumerate USB devices with UDisks: http://mindbending.org/en/python-and-udisks-part-1
Seems simple enough, but I think we should still use our current code as fallback if the UDisks service isn't available/running. It's proven fairly reliable in the past.

Ghostkeeper, 2018-02-05:
Our project manager just removed this from our planning because it's older than 12 weeks.

Dawoodoz, 2020-05-01:
On Manjaro (Arch), Cura 4.5.0 just saves the file to a random internal SSD without asking. I think it's better to fix the export menu to detect external devicas and then remove the quick button that will probably never work on Linux.

MariMakes, 2023-06-01:
Hey @lokster,

Nice to meet you, I'm a fresh set of eyes 👀 
Thanks for the update and for showing us your workaround.
Hopefully, that can help others who are stuck as well. 

I showed them your workaround, and they had a quick look at your code. They concluded that this might work on your device but it is probably not going to work for a wide and diverse set of Linux Distros. 
We believe we will not work on this anytime soon and will therefore close this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I'll close this issue, I hope you understand.

closed 2023-06-01 by MariMakes