Ultimaker/Cura #2708: Imported profiles not showing

magnumuk, 2017-11-01:
Cura 3.0.4 Mac OS x Sierra.
I have tried to import 4 Cura profiles for a model aircraft from Lab Print.
After importing I receive a message advising that the import is successful, but on viewing at my profiles, none are  showing.
Any ideas?
Regards
David
Dr. D. Wilson
Durham 
UK
[image]
[image]

ChrisTerBeke, 2017-11-01:
Hi. Cura only shows the profiles that are compatible with the current machine/material/extruder setup. Do the profiles you imported match those settings?

We might wanna change this behaviour in the future as it's confusing, but for now that's how it is.

magnumuk, 2017-11-01:
Hi Chris,

Wow, a very quick reply, thanks.

The files are Prusa i3 compatible, and I am using an i3 clone.  I have attached one of the profile files for your viewing.  Please let me know what you think.

Regards ..... David
--
Dr. D.  Wilson
The Bungalow
Sunniside
Bishop Auckland
DL13 4LP
Tel: 01388-731466

ChrisTerBeke, 2017-11-01:
The attached files don't seem to come in via email reply, please try using the GitHub web interface to upload these.

Next to the selected printer, also the selected material (and possibly nozzle size / print core) are filtering the shown profiles.

magnumuk, 2017-11-01:
Hi Cris,

Here is the file I attached:

[CURA_thick.txt](https://github.com/Ultimaker/Cura/files/1434425/CURA_thick.txt)

Ellecross, 2017-11-01:
@ChrisTerBeke  It looks like it's an existing issue since 2.6.2 see #2117

ChrisTerBeke, 2017-11-01:
Seems that that issue is still in the process of being fixed. We'll see what happens there.

magnumuk, 2017-11-01:
Thank you Chris,

Yes I also tried to upload these to v2.6.2 without success.
I will have to await an update.

regards
David

magnumuk, 2017-11-01:
Cris,
I have just tried Z3RO11's workaround, and it also works using Cura 3.0.4

    Import the legacy profile into Cura 2.5
   Export profile
    Install and load v3.0.4 and import the new Cura profile we just exported
At least this is a workaround until a fix is sorted.

ChrisTerBeke, 2017-12-01:
Fixed by https://github.com/Ultimaker/Cura/commit/7b4cb1124082854a39d201050edb48ece00f1b9d. This will be in version 3.1 which is scheduled for release next week.

closed 2017-12-01 by ChrisTerBeke