Ultimaker/Cura #2709: Layer height slider missing 0.1

S3D-r, 2017-11-01:
Cura 3.0.4.  Printer: BCN3D Sigma.

Layer height slider is missing 0.1 and has 0.2 instead.

[image]

ChrisTerBeke, 2017-11-01:
The slider shows the layer heights for all available quality profiles (same as old dropdown). The 0.1 layer height quality must have somehow gotten a 0.2 layer height (typo?). Also, how did you create the Sigma machine?

S3D-r, 2017-11-02:
OK, is that typo printer model specific or is it in the Cura code?   

I had earlier "BCN3D Cura 1.0.2 beta" installed. When I installed this Cura 3.0.4. and I first time opened it, there was only this Sigma profile!

ChrisTerBeke, 2017-11-02:
Ah, then the profile actually came from the BCN Cura version. In that case we can't really support that here as it's an issue in their version of Cura (actually they should save user data to a different location on disk if you ask me).

closed 2017-11-02 by ChrisTerBeke

fieldOfView, 2017-11-02:
The solution is probably fairly similar to this:
https://github.com/Ultimaker/Cura/commit/e897431c959491a8cf27fbf275bd2bcabdb15d01#diff-d9718dd9a2eecf1fe13c187b85da7e36
ie: remove the ```"layer_height"``` property from the definition, and instead set the ```"preferred_quality"``` to ```"\*Draft\*"```

ChrisTerBeke, 2017-11-02:
I'll post this issue thread on the BCN3D Cura repo.

ChrisTerBeke, 2017-11-02:
Ref: https://github.com/BCN3D/Cura/issues/17