Ultimaker/Cura #2711: Printer's (Sigma R17) figure incorrect

S3D-r, 2017-11-01:
View shows bed figure of a delta printer, not Sigma's.

Preferences - Printers  shows Printer type: 101Hero.   Don't know, how could that be changed.

[image]

[image]

ChrisTerBeke, 2017-11-01:
As the BCN Sigma is not shipped by default I assume this is a Custom FDM printer or a machine loaded from a project file? If so, the machine settings which contain fields for printer type and build plate model are incorrect.

ChrisTerBeke, 2017-11-02:
Closing because like the other issue it's a result of BCN Cura doing weird things with saving user data and settings.

closed 2017-11-02 by ChrisTerBeke