Ultimaker/Cura #2714: Cura Version 304 can not Output the  Slicing G Code

Hardwarekiller, 2017-11-03:
Cura Version 304 can not Output the Gode !

When you go to the  "Save to File"
give NOT the GCode Output !

Cura Version 303 is OK 
Cura Version 304 can NOT Save a GCode 
This ist a big BUG !

Here the Picture of the BUG from Cura 304!
I can NOT Save an GCODE, 
he is missing

[image]

An here the Cura Version 303 it works

[image]

By 

Hardwarekille

Ghostkeeper, 2017-11-03:
Could you go into the application menu: "Plugins" -> "Installierte Plugins...", and look there if the plug-in "G-Code-Writer" is enabled?

Hardwarekiller, 2017-11-04:
Hi, now it work,s 
thanks for you Tip !

The Problem from Version 304 is, this Version Change Plugin  "G-Code-Writer" automatic  to "off"   after an new installing.
or when i switch between Output to "print via usb" or "save to File" go to  
"G-Code-Writer"  to "off" and not return to "on"

I Thing the Version 303 with name  "Cura 3.0.3" sets the   "G-Code-Writer"  to standart   "on"
the last Version 304 Name with name   Ultimaker Cura 3.0.4 sets the  "G-Code-Writer" to standart   "off"

Thanks 

Hardwarekiller

Ghostkeeper, 2017-11-10:
Weird. I haven't seen that happen.

Some other people have been reporting random plug-ins disabling sometimes. Myself I've been seeing the "Do you want to share anonymous data" box come back and the window position reset. But it's all pretty hard to reproduce :(

EldarPolitkin, 2018-02-16:
Have the same issue

but I cant enable the G-Code-Writer plugin =\

[image]

thenk you,
Eldar/

Ghostkeeper, 2018-03-09:
A workaround is to go to Help -> Show Configuration Folder. There open the `cura.cfg` file with a text editor and under the `[general]` category remove the `GCodeWriter` plug-in from the `disabled-plugins` list. Or just remove the `disabled-plugins` line altogether (which will reset the disabled/enabled state of all your plug-ins).

Ghostkeeper, 2018-03-09:
We don't know how this happens. I haven't seen it occur. As long as we don't know how to reproduce the issue, there is little we can really do to help...

ianpaschal, 2018-05-28:
Being unable to reproduce or fix this, and also the fact that the plugin browser has replaced this UI anyway, I'm going to close this.

closed 2018-05-28 by ianpaschal