Ultimaker/Cura #2715: Cura should depend on Python 3, not 3.5.0

onitake, 2017-11-03:
Cura and the test suite have a hard dependency on Python 3.5.0, which is updated occasionally.
This causes problems when multiple Python interpreters are installed on a system, and especially when a system has a newer Python version available.

CMakeLists.txt and cmake/CuraTests.cmake should depend on Python 3 instead, so the main (or latest) Python interpreter is picked up by cmake.

onitake, 2017-11-03:
(Please disregard PR 2716, I accidentally pushed the wrong branch)

nallath, 2017-11-03:
Because of cross platform issues, we have to depend on at least python 3.5. @awhiemstra knows the specifics. From what I remember it had to do with MSVC issues.

onitake, 2017-11-03:
Ok, makes sense.
But does that have to be reflected in the in the Cmakefiles?
Wouldn't a simple notice in the README suffice?

awhiemstra, 2017-11-06:
find_package(PythonInterp 3.5.0) should find any Python equal to or greater than 3.5.0. At least that is what happens on my system, where it mentions
 
> -- Found PythonInterp: /bin/python3 (found suitable version "3.6.3", minimum required is "3.5.0")

So the version here indicates a minimum, any higher version should also be fine. If detecting the Python interpreter fails, that may point to a different issue.

onitake, 2017-11-06:
That's good, but it's still a problem if multiple Python versions are installed.

For example, if a system has Python 3.5 and 3.6, cmake will only find and run tests against 3.5. This will fail when Arcus or Savitar (which are bound to a single Python version) have been built for 3.6 instead.

Ghostkeeper, 2017-11-11:
That's not so much a problem with using a specific Python version but more with mixing up different Python interpreters.

I'd think that CMake's findPython script would find the same Python version every time though. Maybe there is a difference in the way that we call the find script? 3.5.0 instead of 3.5 or something?

onitake, 2017-11-17:
I understand your reasoning with respect to "find at least Python 3.5", but passing an explicit version to `find_package` seems like the wrong solution. While having multiple Python 3 interpreters installed is not common, it is not impossible either.

According to [the documentation](https://cmake.org/cmake/help/v3.0/module/FindPythonInterp.html), there is a way to pass a list of preferred Python versions, by setting the variable `Python_ADDITIONAL_VERSIONS` before calling `find_package`. That is not exactly "find the latest Python version that is at least 3.5", but would be close enough. The downside is that it requires manual updates to that list of Python versions.

I think the best approach here is to not specify a version in the `find_package` call at all, but check `PYTHON_VERSION_MAJOR` and `PYTHON_VERSION_MINOR` for 3.5 or later by hand.

onitake, 2017-11-17:
Here's another try at this.

ChrisTerBeke, 2017-12-01:
@awhiemstra @Ghostkeeper is the new PR good or not?

Ghostkeeper, 2017-12-05:
Well let's keep that discussion to the PR at hand.

Ghostkeeper, 2017-12-05:
I'm still unable to reproduce this. To test, I installed Python 3.6.3 on a computer that already had Python 3.5.2 installed. CMake found it fine:
[image]

Right here in the source code you can also see that it accepts versions newer than the required version: https://github.com/Kitware/CMake/blob/53c6da8b9e549b07996b446981b4fce9577904a5/Modules/FindPythonInterp.cmake#L54
In the code below that you'll find that great care is taken as to the order in which the Python is found if there are multiple installations.

So as far as I know, CMake will find newer versions fine. I'm thinking that it's something in your installation.

onitake, 2017-12-06:
Looks like it still happens on Linux:
```
$ python3 --version
Python 3.6.3
$ python3-config --configdir
/usr/lib/python3.6/config-3.6m-x86_64-linux-gnu
$ ls /usr/bin/python3*
/usr/bin/python3    /usr/bin/python3.5m  /usr/bin/python3.6-config  /usr/bin/python3.6m-config  /usr/bin/python3m         /usr/bin/python3-unidiff
/usr/bin/python3.5  /usr/bin/python3.6   /usr/bin/python3.6m        /usr/bin/python3-config     /usr/bin/python3m-config
```
Running cmake without the patch won't detect the default interpreter:
```
cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON -DURANIUM_DIR=/usr/share/uranium -DCURA_VERSION=3.0.3
-- Found PythonInterp: /usr/bin/python3.5 (found suitable version "3.5.4", minimum required is "3.5.0") 
...
```
I don't fully understand how this FIND_PACKAGE works, but it seems that the part you referenced only sets up the order in which versions are searched.
The detection is further below, and it seems that it's slightly different for Windows than for other platforms.

If I remove Python 3.5 from my system, cmake will detect Python 3.6 just fine. Odd.

closed 2018-03-02 by ChrisTerBeke