Ultimaker/Cura #2718: "Automatically drop models to the build plate" broken

garrett-carson, 2017-11-03:
Each of these models will slice fine alone, but try to have them both on the same plate and the Q-window will float up to layer 3.

Files are `A-Gothic-Window-v1.0.stl` and `Q-Gothic-Rosetta-Window-v1.0.stl` from [Printable Scenery's Cathedral Walls](https://www.printablescenery.com/product/cathedral-walls/)

[image]
[image]
[image]

Ghostkeeper, 2017-11-08:
How do those models look in X-ray view? Anything red?

I can't test it for you unless I shell €20,- on it.

garrett-carson, 2017-11-08:
No red. 2 Models in question sent to your email listed in cura's git commits.

[image]

Ghostkeeper, 2017-11-09:
It looks like this issue is a duplicate of https://github.com/Ultimaker/Cura/issues/2050.

The Rosetta Window file doesn't get aligned to the build plate properly. I've tried fixing it up in Meshlab and in CloudCompare but none of the tricks that I pulled could get Cura to align it properly (it did find duplicate faces and a non-manifold edge though, but that didn't seem to matter to Cura).

I suggest that you wait for now until we have a fix for #2050. A workaround is to move the Rosetta to Z=-0.166 using the move tool. I measured that your model was 0.166mm above the build plate.

Ghostkeeper, 2017-11-10:
It seems these are all duplicates of each other. Maybe there could be some more information in the other ones:
* https://github.com/Ultimaker/Cura/issues/545
* https://github.com/Ultimaker/Cura/issues/1101
* https://github.com/Ultimaker/Cura/issues/1487
* https://github.com/Ultimaker/Cura/issues/1527
* https://github.com/Ultimaker/Cura/issues/1979
* https://github.com/Ultimaker/Cura/issues/2050
* https://github.com/Ultimaker/Cura/issues/2296
* https://github.com/Ultimaker/Cura/issues/2718

closed 2017-11-10 by Ghostkeeper

Ghostkeeper, 2017-12-15:
This is fixed for Cura 3.2.