Ultimaker/Cura #2719: Cura trims comments out of customized start/end gcode on restart.

filipgoc, 2017-11-03:
I am adding some custom nozzle wiping code to my start gcode. Trouble is, when Cura is restarted, 

1. not only the customized start gcode is not retained 
2. but the whole gcode script is **trimmed** out of all comments other than the 1st one. 
       - the pattern of messing up the code may be more complex, but this is what I saw based on about 8 tests I run. 
2. If the gcode is left untouched (same as json), then nothing happens
2. This is a new issue. I have been an equivalent default gcode, and different versions of custom gcode including a lot of white space and comments since Cura 2.1.

## Test Reproduce 01: 
Go to machine settings and replace the gcode with a custom (test) one
### before
```
; ---------------------------------------
; ; ; Jellybox Start TEST 01 ; ; ;
; _______________________________________
; comment
; comment
G28                   ; home all axes
; comment
G28                   ; home all axes
G28                   ; home all axes
; comment
; comment
G29; 
```

Slice something. 
The code is correctly used in the resulting gcode. No problem here. 
Then, restart Cura. 
Go to machine settings and look at the gcode

### after
```
; ---------------------------------------
G28                   ; home all axes
G28                   ; home all axes
G28                   ; home all axes
G29;
```

## Test Reproduce 02

### before
```
G28                   ; home all axes
; comment
G28                   ; home all axes
G28                   ; home all axes
; comment
; comment
G29; 
```

### after
```
G28                   ; home all axes
G28                   ; home all axes
G28                   ; home all axes
G29;
```

## environment
Mac Sierra
Cura 3.0.3

Ghostkeeper, 2017-11-08:
You say that you've had it working since 2.1, but this is a duplicate of https://github.com/Ultimaker/Cura/issues/1237.

closed 2017-11-08 by Ghostkeeper

filipgoc, 2017-11-09:
You're absolutely right @Ghostkeeper!

I have now re-tested it with Cura 2.3.2 and indeed it's the same issue. 

It's an official duplicate. 

Not sure where I made the mistake to cause me to think it used to work. I did all proper testing in Cura 3, must have mistakenly left only the json parsed scripts in older versions.