Ultimaker/Cura #2721: Cura 3.0.4 Linux won't read printer definitions

F1ynn, 2017-11-03:
Using Cura 3.0.4 on Ubuntu 17.10, I placed the machine definitions for my MP Select Mini (files attached) in ~/.local/share/cura/3.0/definitions. When attempting to add a new printer, Cura appeared to not see them. Placing the same files in the applicable folder on the latest Windows version of Cura worked fine.
[monoprice_select_mini.zip](https://github.com/Ultimaker/Cura/files/1441757/monoprice_select_mini.zip)

ChrisTerBeke, 2017-11-04:
Are you sure the definition file is correct? I think the version number should be 3 and preferred quality profile should be a quality type instead of a layer height. Also the quality profiles should be in the qualities folder, not the definition folder.

F1ynn, 2017-11-06:
After making those two changes, I still don't see the printer showing up in the list. I am sure there are errors in the definition, but at least in the Windows version of Cura it loaded.

ChrisTerBeke, 2017-11-06:
Mmm there shouldn't be any difference between operating systems indeed. All other settings are the same between your 2 installs?

F1ynn, 2017-11-07:
Yes, all settings are at their default values on both installs.

Ghostkeeper, 2017-11-14:
No, the version number needs to be 2: https://github.com/Ultimaker/Cura/blob/913bcf45b184fe01392e41803a260d37b333a30c/resources/definitions/ultimaker2_plus.def.json#L3

Ghostkeeper, 2017-11-14:
The preferred quality and variant match on (user-readable) profile names. Hence the asterisks. Take for instance the Ultimaker 3 preferred quality:
https://github.com/Ultimaker/Cura/blob/913bcf45b184fe01392e41803a260d37b333a30c/resources/definitions/ultimaker3.def.json#L20

This matches on the `Normal` with capital N, not on the `normal` quality type.

Ghostkeeper, 2017-11-14:
When I put the definition file in my source directory in `.../Cura/resources/definitions` then it loads the definition fine. It seems that the definition folder is not loaded at all on Linux then.

ajford, 2017-12-02:
I was running into the same issues. Running the Cura-3.0.4.AppImage, on Ubuntu 17.10 (64 Bit). 

When I put 

    "preferred_quality": "*Normal*", 

into my `monoprice_select_mini.def.json`, I was then able to pull up the new definition. Before that, no luck.

tristan-shelton, 2017-12-29:
I'm having similar troubles here with Cura 3.1.0 appimage on Mint (an Ubuntu variant). I put the .def.json in ~/.settings/cura/settings.definitions/ per [the instructions](https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura) and [the data paths](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations) on the wiki.

Unfortunately changing the preferred quality setting to `*Normal*` didn't do the trick for me. :(

ajford, 2017-12-29:
@tristan-shelton, check your paths again. On linux, the path that worked for me (and is specified in the wiki) is `~/.local/share/cura/VERSION/settings/definitions/`, where `VERSION` is most likely `3.1.0` or `3.1`. 

You specify `~/.settings`, which could be a typo in your comment, but just double checking. 

Also, to quote Roy from IT Crowd, "have you tried turning off and on again?" When I tried things, it wouldn't register the file until after a restart.

closed 2018-11-26 by Appesteijn