Ultimaker/Cura #2722: 3.0.4. Cura per part settings infill

sierradays, 2017-11-03:
Infill settings not working.  100% will fill, but anything less is not correct

smartavionics, 2017-11-03:
Hi, could you please provide more info - screenshots? model? settings? Without that, it is impossible to understand your problem.

sierradays, 2017-11-03:
Here is a view from the top, this is between two walls.  You can see the infill pattern doesnt fill the area or bridge to the next wall.  Infill Setting are 20% grid infill.

[image]

smartavionics, 2017-11-03:
Yes, that looks weird. Could you please post a link to the model or upload it so I can try slicing it myself.

smartavionics, 2017-11-04:
Just a thought, make sure that gradual infill steps is set to 0. Maybe that is non-zero and doing something weird?

smartavionics, 2017-11-04:
Also, in the above image, where is the bottom skin? The skirt is visible so you are displaying all of the layers but there's no skin under the weird infill.  Can you please show a similar image but this time use the layer view so that the various components (skin,  infill, etc.) are colour coded.

closed 2017-11-05 by sierradays

Ghostkeeper, 2017-11-10:
When you set infill to 100% we automatically set the number of bottom skin layers to 999999. This is because it should have the same effect but prints faster, achieves better infill density and better quality wall adhesion. The downside is though that your preferred infill settings won't work any more then. A workaround is to set the infill to 101% (or the bottom layer count back to 5 or so).