Ultimaker/Cura #2723: Unable to import 2.5 Profiles

AndriesMuylaert, 2017-11-04:
Hello,

I'm unable to import old ini profiles in Cura 3.0.4.
[Profiles.zip](https://github.com/Ultimaker/Cura/files/1443126/Profiles.zip)

ChrisTerBeke, 2017-11-04:
This is a known issue and we're working on resolving it.

AndriesMuylaert, 2017-11-04:
Ok thanks,

Meanwhile I found a workaround by installing 2.5, importing there,
exporting with new fileformat, reinstalling 3.0, importing the new
fileformat.

Best regards,

Andries

nickthetait, 2017-11-08:
Duplicate of https://github.com/Ultimaker/Cura/issues/2117
@ChrisTerBeke can you comment on the main issue that this is a WIP, don't want someone wasting their time trying to fix this if Ultimaker already is... 
Also, hooray that this will be taken care of! This (currently broken) feature is a major hurdle for people upgrading to the Qt version of Cura.

ChrisTerBeke, 2017-11-08:
Done. Closing this since it's a duplicate.

closed 2017-11-08 by ChrisTerBeke