Ultimaker/Cura #2724: Stop auto-creating new printers

thesec, 2017-11-04:
I now have 8 printers while I did not add or change the printers myself.
All are the same, only the label is added "#n" as a suffix.

Please stop automatically creating new printers.

ChrisTerBeke, 2017-11-04:
This usually only happens when you open project files and choose "Create new" instead of "Update current" for the saved printer information. What is your workflow that creates these new printers?

thesec, 2017-11-04:
Thanks @ChrisTerBeke just realised this the other minute. Yet there is no different printer in the project and still the prompt create/update comes up?

Also, I do not see the use to update e.g. a Delta with any cartesian printer settings and vice versa.
Seems easier and a lot less prone to errors when loading the model and its slicer settings standalone, clearlier separating concerns ...?

ChrisTerBeke, 2017-11-04:
Project files are intended to store everything so you can share then with colleagues etc. The popup only allows updating though when the printer types are the same (so a Delta will never become a UM3 or something). The popup will show the create/update prompt if you have a similar printer selected because you might have specific extruder/material settings attached to it (as it stores the whole settings stack).

thesec, 2017-11-04:
Too much "magic" under the hood to me, too many concerns mixed up.

To me the slicer is only the piece getting the model prepared for printing.
The printer does not matter for it to do the job or to allow sharing the settings with other, especially when they have another printer.

ChrisTerBeke, 2017-11-04:
Material and printer settings are very important to generating good quality slicer output. Sharing all this information in project files is very important to office/professional use cases, which is the biggest user group for FDM style printers nowadays. I don't see any change in this functionality anytime soon. Sorry if you don't like it that way. You can also share 3D models or quality profiles separately (import/export from the profile manager preferences window).

closed 2017-11-04 by ChrisTerBeke

thesec, 2017-11-04:
None mentioned material before.
Printer's settings are not required for a slicer's output. Only the measurements are of interest - but still no must-haves.

ChrisTerBeke, 2017-11-05:
Actually a whole bunch of settings are important for slicing. Take the UM3 definition profile for example (https://github.com/Ultimaker/Cura/blob/master/resources/definitions/ultimaker3.def.json), there's over 60 settings in there that Cura uses to optimise the G-code. Sure, there's printers with less settings, but that usually shows in the print quality...

thesec, 2017-11-05:
Important for slicing yes, but the least amount is defining the actual printer.
All but e.g. print volume, heated bed or none, number of extruders are set up per "slice" for professional results.

But this is getting philosophical and I do really not want you guys to change that config part of your overall excellent and much appreciated piece of software.
Kudos for taking your time.

Ghostkeeper, 2017-11-10:
> office/professional use cases, which is the biggest user group for FDM style printers nowadays.

That's not true, actually. I don't know where people keep getting that idea from other than echo chambers, but we have quite clear data from the form on Cura's download page that suggests otherwise.

Back on topic...

> Seems easier and a lot less prone to errors when loading the model and its slicer settings standalone, clearlier separating concerns ...?

The whole function of workspaces (or "Cura projects") is to save the settings along with the model for easy reproduction of a print. You can export the settings separately as a profile or export the model separately if you want them separate.

Sometimes it may matter a lot what printer definition you're using, such as when trying to reproduce a dual extrusion print on a single-extrusion machine. Sometimes it will not matter a whole lot. In the latter case we probably wouldn't need to create a new printer in your Cura installation, to people that aren't too picky about it. But we haven't built the logic to discern whether it's important. It's a bit of a fuzzy border.