Ultimaker/Cura #2725: Simplify Profile Management with Multiple Extruders

thesec, 2017-11-04:
As is profiles store all settings for all extruders.
As such it prevents easily taking over the settings applied to e.g. the left to the right extruder and vice versa.

Already being outvoted on a checkbox to automatically apply all settings to either extruder ... I suggest to save the settings independently of the extruder/s and apply them on load to the currently selected one.

That would not only help a ton in multi-extruder setup but would also be more intuitive as what happens happens in context and not underneath the hood (applying invisible settings in other tabs).

ChrisTerBeke, 2017-11-04:
Settings that are tied to an extruder are stored with that extruder (also in project files). You can right click on a setting and select "Copy to other extruders" to synchronise that setting.

Settings are already stored separately from an extruder (in different setting containers behind the scenes). Any extruder slot (0, 1, 2. etc) can be outfitted with a variant (print core or extruder), material, quality settings and user quality changes.

Given this, I'm not really sure what end-user functionality you want to add. Can you give a specific user scenario of what you imagine to be a better setup?

thesec, 2017-11-04:
Copying them over one by one is quite tedious.

The use case that I stumble over is: Set up a process and finetune it on one extruder and get it over to the other.
I tried that setting up everything on the first extruder, saving the profile (which happens in the context of this very single extruder's tab!). I then loaded another model and project, switched to the second extruder and loaded the new profile just to see that nothing in its settings changed.

What I'd very much like to be able to is to save the custom print setup (underneath its label the profile are listed ... to me the perceived context ...) on any extruder, switch to another extruder, load the stored custom print setup and see all settings applied to it.

There is no single slicer on the market atm that has this and is something I stumble over every now and then printing my models on different machines using the same filament but different colors or just slightly different material that I'd like to inherit the working settings for.

Ghostkeeper, 2017-11-10:
Duplicate of #2448

closed 2017-11-10 by Ghostkeeper