Ultimaker/Cura #2726: Object floats if support is on

ZerialLim, 2017-11-04:
Support structure off : fine

Support structure ON :
The whole object lifts for ONE layer, causing
- lack of bed adhesion
- no brim structure for objects
- failure to make flat bottom surface
but the support structure starts from layer 1, causing
- any brim or skirt will be overlap with the object
- cannot just ‘move my Z axis’ to correct the issue.

Ghostkeeper, 2017-11-10:
This is most likely caused by the model not being positioned exactly on the bed.

If support is off, the first layer is completely empty. Our slicer back-end recognises this and moves the rest of the layers down then so that it all gets put nicely on the build plate. When you enable support, the first layer is no longer empty so it doesn't do that.

We've seen a bug like this where Cura thinks it's positioned on the bed properly but it's actually floating.
See one of these issues:
* https://github.com/Ultimaker/Cura/issues/545
* https://github.com/Ultimaker/Cura/issues/1101
* https://github.com/Ultimaker/Cura/issues/1487
* https://github.com/Ultimaker/Cura/issues/1527
* https://github.com/Ultimaker/Cura/issues/1979
* https://github.com/Ultimaker/Cura/issues/2050
* https://github.com/Ultimaker/Cura/issues/2296
* https://github.com/Ultimaker/Cura/issues/2718

closed 2017-11-10 by Ghostkeeper

Ghostkeeper, 2017-12-15:
This is fixed for Cura 3.2.