Ultimaker/Cura #2728: Holes disapearing after slicing

Peterzij, 2017-11-04:
I made a simple 90 degree angle bracket. STL downloaded in Cura (using Fusion 360 to draw it) and in the solid view I can see all the holes. But going to layer view the holes are gone. 
[image]
[image]

Thisismydigitalself, 2017-11-04:
check to see if you have "Remove all holes" checked..

fieldOfView, 2017-11-05:
If that does not work, search for the setting "Union Overlapping Volumes" and turn it off. If that helps, your model has issues.

Peterzij, 2017-11-06:
My God.. That it can be so simple... Thanks.. I never thought about that. Thanks Guys

Thisismydigitalself, 2017-11-06:
Not your fault.
I love Cure.  Cura is a very powerful slicer but this comes with a price. much too often I can't see the wood for the trees as they are so many settings to play with and much too often an hidden setting can and will have negative effect on the finished product and I find myself going back and forth between layer view and solid view for making sure that WYSIWYG . I proposed to the Cura team to add favorites or some color mapping for settings which we use most for not getting lost in the forest. like filters for profiles.

closed 2017-11-06 by ChrisTerBeke

dunk7, 2022-10-27:
Thx for info, but for future reference for anyone else: Make sure you have your face normals flipped the right way

dunk7, 2022-10-27:
oh actually I was using blender and I guess you have to leave your cavities as separate objects and doing the boolean modifier messes up the mesh or something.