Ultimaker/Cura #2730: Cura 3.0 Not actually changing print settings.

Mav68erick, 2017-11-05:
When open for to long, I notice changing anything in your profile, although it may change on the screen and you save it. Sometimes doesn't actually change it. I can have nozzle temp at 215. Save my gcode and load it. and its back to 200 or 205. I can visually notice when this isn't working by changing my infil %. turn t to 1% or to 100% and it visual makes no changes on the preview. When you close cura and re open it, its back to the same invisible settings it reverts too.

Sometimes when you save a current profile. and select a different one, it automatically changes the one you've switched to, to the same as the last profile you are in,

Vandrasc, 2017-11-09:
Hi @Mav68erick, which version are you using? 3.0.3 or 3.04?

Mav68erick, 2017-11-09:
I will check when I get home. I haven't updated to the very latest. It's
whatever the current version was 2 weeks ago

Mav68erick, 2017-11-09:
Im using 3.0.3 Its was the latest when this happened. No matter what
profile settings I changed, it would change on screen, I would slice, and
save the gcode. Run it on my printers, it would default back to something
else.

nallath, 2017-11-15:
Could you post your logs and a project file?

konvoj, 2017-11-16:
The issue can be that you are changing the settings for the second extruder. It happened to me several times that the object was set to different extruder that i was changing in the setting. I already reported the issue.

Mav68erick, 2017-11-16:
Ok. I did accidentlt turn multiple extruders on, and turned them off. And
haven't noticed an issue yet. But I do close cura dialy

Ghostkeeper, 2017-11-25:
Let's assume that was it then. This is quite a common problem. Is there anything we can do to make it more clear that you're editing on the wrong extruder?

Mav68erick, 2017-11-25:
I remember when extruder 1 2 3 tabs were at the top slightly.  How about
slightly change the background of each extruder to different colors? Or
different shades, that people can edit as they wish.

I have yet to have this issue happen again now.

The only other issue is switching profiles and even if I say keep or
discard, it will sometimes change a few of the settings. And I have to go
back and say info. I have to update that profile before selecting another
faithfully or if you wait for the pop up menu it can mess up

closed 2018-03-09 by ianpaschal

Ghostkeeper, 2018-04-04:
They now have different outline colours, but I think that might already have been the case when this issue was reported.