Ultimaker/Cura #2731: Cura.Exe has stopped working when trying to save Gcode

marchantricky, 2017-11-05:
Only Cura 2.6.2 will work for me.  All the new ones that I download seem to accept the STL and slice it just fine.  When I try to "Save to File" or "Save to removable device", it will save a 0kb Gcode and then I will get that error message.  How can I fix this issue!???

ChrisTerBeke, 2017-11-05:
Can you share the logs with us? The readme in this repository has information on how to find them.

marchantricky, 2017-11-05:
2017-11-05 11:10:07,869 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-11-05 11:10:08,085 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
2017-11-05 11:10:08,127 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
2017-11-05 11:10:08,176 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeReader
2017-11-05 11:10:08,205 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeWriter
2017-11-05 11:10:08,267 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ImageReader
2017-11-05 11:10:08,321 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LayerView
2017-11-05 11:10:08,368 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2017-11-05 11:10:08,421 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-11-05 11:10:08,488 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2017-11-05 11:10:08,651 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OctoPrintPlugin
2017-11-05 11:10:08,705 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
2017-11-05 11:10:08,721 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2017-11-05 11:10:08,767 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
2017-11-05 11:10:08,820 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PostProcessingPlugin
2017-11-05 11:10:08,889 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-11-05 11:10:08,951 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-11-05 11:10:08,989 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SolidView
2017-11-05 11:10:09,051 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-11-05 11:10:09,120 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-11-05 11:10:09,189 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-11-05 11:10:09,252 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-11-05 11:10:09,305 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-11-05 11:10:09,368 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-11-05 11:10:09,453 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UltimakerMachineActions
2017-11-05 11:10:09,638 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-11-05 11:10:09,706 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UpdateChecker
2017-11-05 11:10:09,722 - INFO - UpdateChecker.UpdateCheckerJob.run [44]: Checking for new version of cura
2017-11-05 11:10:09,853 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin USBPrinting
2017-11-05 11:10:09,906 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
2017-11-05 11:10:09,990 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2017-11-05 11:10:10,037 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2017-11-05 11:10:10,106 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2017-11-05 11:10:10,168 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2017-11-05 11:10:10,237 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2017-11-05 11:10:10,306 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2017-11-05 11:10:10,388 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2017-11-05 11:10:10,469 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
2017-11-05 11:10:10,538 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
2017-11-05 11:10:10,538 - INFO - UM.Qt.QtApplication.__init__ [114]: Command line arguments: {'external-backend': False, 'file': [], 'single_instance': False, 'qmljsdebugger': None, 'disable-textures': False}
2017-11-05 11:10:10,592 - INFO - UM.VersionUpgradeManager.upgrade [112]: Looking for old configuration files to upgrade.
2017-11-05 11:10:10,592 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening
2017-11-05 11:10:11,054 - INFO - UM.Backend.Backend.startEngine [75]: Started engine process: C:\Program Files\Ultimaker Cura 3.0\CuraEngine.exe
2017-11-05 11:10:11,054 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 3.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Ultimaker Cura 3.0\\resources\\definitions\\fdmprinter.def.json', '']
2017-11-05 11:10:11,339 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-11-05 11:10:11,339 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Cura_SteamEngine version DEV
2017-11-05 11:10:11,339 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Copyright (C) 2017 Ultimaker
2017-11-05 11:10:11,355 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-11-05 11:10:11,355 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] This program is free software: you can redistribute it and/or modify
2017-11-05 11:10:11,355 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] it under the terms of the GNU Affero General Public License as published by
[... cut ...]
2017-11-05 11:21:17,671 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.0\resources\shaders\striped.shader]...
2017-11-05 11:21:17,704 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.0\resources\shaders\default.shader]...
2017-11-05 11:21:17,724 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.0\resources\shaders\grid.shader]...
2017-11-05 11:21:17,748 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 3.0\resources\shaders\platform.shader]...

marchantricky, 2017-11-05:
this is it.  I installed it again, and tried again.  Check it out and let me know what you think.

LipuFei, 2017-11-08:
@marchantricky May I ask which Windows version are you using? and does it happen to any STL model or just some specific ones?

Ghostkeeper, 2017-11-14:
> 2017-11-05 11:19:34,061 - DEBUG - LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [136]: Writing to Local File C:/Users/march/Desktop/Auto Desk Files/STL/gcodes/asdfasdf.gcode in text [...]

Looks like you got a segfault when sending anonymous user statistics (which happens when you save the file). Perhaps you could try turning that off in the preferences and trying again, as a workaround?

I'm unable to reproduce that bug here. It saves fine. However we've heard before that some Windows Explorer plug-ins can cause Qt's file dialogues to crash. In that case, saving to a USB stick should still work so you could try that to see if that was the case. The bug in the file dialogues was caused by the Google Drive extensions and the Adobe extensions.

moninox, 2017-12-25:
Hi,
same problem here. 

I tried 3.04 and 3.1.0  (32&64 bits) and crash happens when I save de gcode file, after slicer stage finish. Gcode file is created but empty.

**System:**
Nombre del sistema operativo:              Microsoft Windows 10 Home
Versión del sistema operativo:             10.0.16299 N/D Compilación 16299
Fabricante del sistema operativo:          Microsoft Corporation
Configuración del sistema operativo:       Estación de trabajo independiente
Tipo de compilación del sistema operativo: Multiprocessor Free
Id. del producto:                          00325-94433-99163-AAOEM
Fecha de instalación original:             02/12/2017, 23:52:03
Tiempo de arranque del sistema:            24/12/2017, 15:26:45
Fabricante del sistema:                    Jumper
Modelo el sistema:                         EZbook
Tipo de sistema:                           x64-based PC
Procesador(es):                            1 Procesadores instalados.
                                           [01]: Intel64 Family 6 Model 92 Stepping 9 GenuineIntel ~1101 Mhz
Versión del BIOS:                          American Megatrends Inc. JUMPER10x.P08.WP313R.NHNAUHL01, 19/08/2017
Cantidad total de memoria física:          2.921 MB
Memoria física disponible:                 774 MB
Memoria virtual: tamaño máximo:            4.073 MB
Memoria virtual: disponible:               1.132 MB
Memoria virtual: en uso:                   2.941 MB
Revisión(es):                              6 revisión(es) instaladas.
                                           [01]: KB4048951
                                           [02]: KB4053577
                                           [03]: KB4054022
                                           [04]: KB4055237
                                           [05]: KB4058043
                                           [06]: KB4054517

**Cura logs:**
2017-12-25 01:45:05,980 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2017-12-25 01:45:06,089 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 3.1\CuraEngine.exe
2017-12-25 01:45:06,089 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2017-12-25 01:45:06,151 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileReader
2017-12-25 01:45:06,198 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileWriter
2017-12-25 01:45:06,245 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2017-12-25 01:45:06,292 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2017-12-25 01:45:06,370 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2017-12-25 01:45:06,417 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2017-12-25 01:45:06,480 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-12-25 01:45:06,872 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
2017-12-25 01:45:06,934 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
2017-12-25 01:45:06,997 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeReader
2017-12-25 01:45:07,044 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeWriter
2017-12-25 01:45:07,108 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ImageReader
2017-12-25 01:45:07,155 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2017-12-25 01:45:07,248 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-12-25 01:45:07,311 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2017-12-25 01:45:07,561 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OctoPrintPlugin
2017-12-25 01:45:07,655 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2017-12-25 01:45:07,717 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
2017-12-25 01:45:07,780 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PostProcessingPlugin
2017-12-25 01:45:07,874 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-12-25 01:45:07,999 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimulationView
2017-12-25 01:45:08,077 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-12-25 01:45:08,155 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SolidView
2017-12-25 01:45:08,217 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-12-25 01:45:08,311 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-12-25 01:45:08,405 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-12-25 01:45:08,483 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-12-25 01:45:08,544 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-12-25 01:45:08,638 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-12-25 01:45:08,779 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UltimakerMachineActions
2017-12-25 01:45:08,997 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-12-25 01:45:09,091 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UpdateChecker
2017-12-25 01:45:09,107 - INFO - UpdateChecker.UpdateCheckerJob.run [34]: Checking for new version of cura
[... cut ...]
2017-12-25 01:47:27,511 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-12-25 01:47:49,557 - DEBUG - LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [118]: Writing to [D:/BQPI3H_110XV6.gcode]...
2017-12-25 01:47:51,548 - DEBUG - LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [136]: Writing to Local File D:/BQPI3H_110XV6.gcode in text mode
2017-12-25 01:47:51,564 - INFO - SliceInfoPlugin.SliceInfoJob.run [29]: Sending anonymous slice info to [https://stats.ultimaker.com/api/cura]...

moninox, 2017-12-25:
Hi again,
In general settings I uncheck "Send print information" and no crash anymore!

Thx.

ianpaschal, 2018-05-29:
Hi @moninox and @marchantricky, are you still experiencing this issue? We're pretty sure we fixed this in the last few months but want to be sure.

If not, no need to respond, if so, please do respond, and I'll re-open.

closed 2018-05-29 by ianpaschal

Ghostkeeper, 2018-07-10:
The crash with sending print information on MacOS will be fixed for the release after 3.4. It's fixed by upgrading to Qt 5.10 which didn't make it into the 3.4 release.

jefhil, 2019-02-14:
this bug -- crashing while saving gcode unless unchecking "send print information" -- still happens in cura 3.6 and 4.0 beta under windows 10

Ghostkeeper, 2019-02-18:
Yes we had to downgrade to Qt 5.9 again because some fairly commonly used MacOS versions were not supported by Qt 5.10 any more. Then we got the bug again, and we fixed it again in a different way but I'm not sure if that was on time for the first beta. I hope that you can try again when the second beta of 4.0 gets released.