Ultimaker/Cura #2733: Automatically close info popovers

thesec, 2017-11-05:
"File Saved ..." and other info only popovers remain on screen, stack above each other and have to be closed manually.
Please have them automatically close after a second or two.

DDDirk, 2017-11-05:
I see them disappear after a few seconds (Cura 3.0, linux). Only the upgrade-popup has to be removed with the mouse.
Not that I actually read them, but in general I don't like popups disappearing while I'm reading them. What I would prefer is that they can be removed with Esc.

thesec, 2017-11-05:
Popovers for info only and not requiring further user interaction like "File Saved ..." should automatically disappear and not keep nagging the user.

Where a user action is required, e.g. with a setting prohibiting slicing that might be handled different.
Overall the same class of information should display and disappear in the same way, keeping users' expectations and interactions consistent across the application's UI.

That said the period might be extended to a few more seconds but should the info-only popovers automatically disappear and not require an extra click or keyboard hit.

DDDirk, 2017-11-05:
Errors and warnings should never disappear automatically.
Esc is very standard and quickly done. For me it's an automatism (which alas doesn't always work).
The check for updates can be disabled in Preferences > General. This should also be possible for those File Saved popups, which I find rather useless

thesec, 2017-11-05:
For a user interaction to be successful you require a clear context and the user (input) focus which popovers do intentionally not receive. That is different for interactive dialogs.

It does not require an extra setting for this standard UI behaviour.

diegopradogesto, 2017-11-09:
The messages we created should automatically close after 30 seconds. I have tested the case in Windows and Mac and the message closes if you wait a bit.
Please let me know if you encounter this behavior again.

exitgames, 2017-11-09:
The bottom most and first stay even longer, other than expected and best practice (users perceive something as stuck at about 10 seconds, so did I).
The times should be decreased and the order in which popovers disappear reversed for (better) UX.

diegopradogesto, 2017-11-09:
There are some messages that never closes till user click on it, such as the collecting data or the upgrade.
I will comment your suggestions about reducing messages times with our UX team.
Thanks for your comments! I will keep you posted.

Ghostkeeper, 2017-11-15:
What did the UX team say about this? I agree that 30 seconds is quite long for most things we display there.

Some messages are infinite on purpose, since we feel that the user should not accidentally miss them by being away from the computer.

diegopradogesto, 2017-11-15:
UX team is aware of this. Probably they need some time to decide, but I will ask them about it.

Ghostkeeper, 2017-11-24:
Maybe we could go the way that it happens in Windows: The messages stay indefinitely until user action is detected (mouse movement or key presses). After that a timer starts running. That would be kind of fancy.

ianpaschal, 2018-02-28:
@LukeChen-Ultimaker This is something to adjust in your notification project.

LukeChen-Ultimaker, 2018-03-27:
To have this specific notification a "10 second" timeout as a start and see how users react to the behaviour. I do suggest that depending on type of notifications, timeout should have differences.

thesec, 2018-03-27:
None but the devs will understand the difference between timeouts for different type of notifications. Users look for consistency. Btw - from about 7 seconds on a system is perceived as stalled when there is no indicator of something in progress.

ianpaschal, 2018-03-27:
Oh the devs understand. Some of us even have master degrees in UX design so we know the general idea at least.

Different times for different types is a nice place to start. But how do you define the types? What happens when you have a new type that doesn't fit in your schema? And then how do you combat the fact that 3 different notifications all have different timeouts when many other users would expect all notifications to have the same timeout (as many notification systems do)? But that's something @LukeChen-Ultimaker is working on.

I will agree though that 10 seconds is waaaaaaay too long. I'd say 5 is about the upper max with 2-4 being a sweet spot.

LukeChen-Ultimaker, 2018-03-27:
The point of defining notification types is to provide a sensible user mental model in terms of notification. 
If all notifications are informing user but not suggest actions to follow up, it wouldn't make sense to design different time out. However, if some notifications do provides follow-up actions ("Open file" button after saving project) we can provide a slightly longer time out. Another question is: how does user end up with multiple notifications at the same screen? Maybe throwing less notification is an interesting direction to make notifications usable?

Anyway, the goal is to provide a very basic framework that we can start iterating on..
@ianpaschal, let's agree on the seconds while we are tweaking the behaviour.

thesec, 2018-03-27:
@ianpaschal "None but the devs" ... ;)

Other than the timing there is plenty of options to provide and indicate follow-up actions ... colors, having a button, different locations, being modal ... just adding in a few more directions of "notification systems".

Throwing less in a basic framework sounds a good start to iterate from. Happy to see you tweaking 👍

ianpaschal, 2018-03-27:
> @ianpaschal "None but the devs" ... ;)

Whooops misread that.