Ultimaker/Cura #2734: 3.0.4 adds projects to the existing one

thesec, 2017-11-05:
Loading a new project file in 3.0.4 adds it to the current one.
Expected: Replace the current project.

Vandrasc, 2017-11-06:
Hi @thesec, I am not able to reproduce.

When I load a new project, the open project pops up asking if I want to update the existing file or Create a new one, works as expected for me. Or the project will be updated or a new machine and project file will be created. 
[image]

I also have the option to load the file as a project file or as a model
[image]
When you choose to import as model, the model will be added to the current project (correct)

Silly question, but did you selected Import as models instead of Open as project?

thesec, 2017-11-06:
Hmm, those screenshotted dialogs do not show to me ...
The way I open a project is: select "File" menu, "Open File(s)", select the 3mf project file.

ChrisTerBeke, 2017-11-06:
You might have a preference enabled to automatically choose one of these behaviours. You can also set the preference to always give the dialog so you can choose.

thesec, 2017-11-06:
Opening and saving files was set to alway import the model - thank you @ChrisTerBeke !

closed 2017-11-06 by thesec