Ultimaker/Cura #2735: 3.0.4 - Backspace does not remove model

nicolinux, 2017-11-05:
On the Mac the backspace key used to remove the selected model from the build plate. This does not work any more with this version.

ChrisTerBeke, 2017-11-05:
It's now working with fn + backspace (as that's the official Mac way to press 'delete').

nickthetait, 2017-11-08:
How about adding backspace (on all platforms) as an alternative way to trigger this action?

ChrisTerBeke, 2017-11-08:
I'm fine with that, but not sure if that's what Mac users expect their software to do (as it's not like that in other parts of the OS).

Ghostkeeper, 2017-11-14:
We explicitly added that in 2.1 because it's easier to use: https://github.com/Ultimaker/Cura/commit/a7d43efe1523292958a91fde1bf095fb635c6881

Ghostkeeper, 2018-02-05:
Our project manager removed this feature from our planning because it was older than 12 weeks. Or rather, it would be older than 12 weeks after one week.

Ghostkeeper, 2018-02-05:
Seems this is fixed though.

nicolinux, 2018-02-05:
I still have this problem with the 3.2 beta (mac). I have an external keyboard. The issue pops up sometimes so further tests are needed. I can reproduce it yet. It seems to depend on some UI state Cura is in when it does not work.

ianpaschal, 2018-03-02:
This is now fixed thanks to @Ghostkeeper.

closed 2018-03-02 by ianpaschal