Ultimaker/Cura #2736: 3.0.4 - One at a time and infill mesh

nicolinux, 2017-11-05:
When the print mode "one at a time" is enabled, and one wants to define an object as an infill mesh, then Cura should ignore the "bounding box" of that infill mesh. Reason for that is that the bounding box of the infill mesh might be so big as that Cura will refuse to slice if it expands outside of the building plate. But since it is "just" an infill mesh, the bounding box should not be regarded in this case.

On the following screenshot you can see that I wanted to use a small box in order to reinforce one area of a frame temple. This is not possible here if the print mode "one at a time is selected".
[image]

diegopradogesto, 2017-11-09:
I have checked and Cura says that is unable to slice because one of the models is outside the buildplate, but it's showing this message because the two models are overlapping, so the message is also incorrect.
[image]

Yes, this is something we need to take a look at. I wanted to suggest that you can create a group with the two models but I noticed that CuraEngine crashes.

diegopradogesto, 2017-11-09:
Devs, take a look at CURA-4551

closed 2023-06-13 by github-actions[bot]