Ultimaker/Cura #2737: 3.0.4 - No post processing cgode feature available

Raffa2s, 2017-11-05:
Hi, 
I just have installed the lastest version of Cura (version 3.0.4)  on my W10 pc, but i can't run Post Processing option under Extension :(
The botton "Modify G-Code" is unclickable (it doesn't turn blue when I pass on it), what can i do? I've just uninstalled and clean my whole pc and nothing changed.
Thanks

nallath, 2017-11-08:
Post your log files please.

Raffa2s, 2017-11-09:
This is my log file, i've tried to manually add the PostProcessing folder but nothing works.
`
[cura.log](https://github.com/Ultimaker/Cura/files/1457150/cura.log)
`

diegopradogesto, 2017-11-09:
I tried the same with your same configuration and I can use the postprocessing plugin. As you can see in the picture 
[image]
, the button is enabled.

I checked the log and the weird thing is that after you upgrade the Cura version, the PostProcessingPlugin is trying to load twice at start-up.
Could you please try to deactivate the plugin, restart Cura, and then activate it again (and restart again). You can do it at Plugins -> Installed plugins.

Send the logs after that, please.

Raffa2s, 2017-11-09:
Hi,
i've tried to deactive one of those two and nothing come from it :( what can i do? i also have unistalled and reinstall the software 
[image]
this is teh log
[cura.log](https://github.com/Ultimaker/Cura/files/1457591/cura.log)

Raffa2s, 2017-11-09:
i have also tried to cut the folder "PostProcessing" and now i have only one tick in Plugins
[image]
here the log file
[cura.log](https://github.com/Ultimaker/Cura/files/1457620/cura.log)

diegopradogesto, 2017-11-09:
For some reason you got several copies of the plugin and there were some conflicts.
Are you able to access the Modify GCode in extensions now?

Raffa2s, 2017-11-09:
no :(

Ghostkeeper, 2017-11-15:
Maybe you could take a look here: `C:\Users\<Raffa2s>\AppData\Roaming\cura\3.0\plugins\`. Is there a PostProcessingPlugin folder there? If there is (somehow) then that could cause the plug-in to load twice.

Why in hell that would disable the menu item, I wouldn't know though.

closed 2018-03-02 by ChrisTerBeke

russgraus, 2018-03-29:
having the exact same issue with cura 3.2.1

fieldOfView, 2018-03-29:
Try selecting the menu item with your cursor keys, and then select the item with the enter key.

russgraus, 2018-03-29:
Thank you that worked

luckylad99, 2018-04-27:
Have the same problem in 3.3.1 and 'filedOfView's above solution worked .... thanks.