Ultimaker/Cura #2738: Cura 3 Release Notes Do Not Specify To Which Version The Notes Apply

mbmast, 2017-11-06:
I have Cura 3.0.3 installed.  When I go to the download page, I see Cura 3 is available for download and I see the release notes.  When I view the release notes, it says it's for Cura 3.0.  Well, I already have 3.0.3 installed, so I must have the most recent version, right?  Wrong.  When I downloaded Cura 3 (and I already have version 3.0.3 installed), I see that it's version 3.0.4, which is NEWER than 3.0.3.  If I just relied on the release notes, I would not have downloaded and installed the newer version.

Release notes must state the complete version number, not just the first part of the version number.

Ghostkeeper, 2017-11-10:
Yes, we forgot to add the 3.0.3 notes to the 3.0.3 release. It should be in 3.0.4 though:

https://github.com/Ultimaker/Cura/blob/41acedd4c13bbb5733a4b4f422e46d1752a696d9/plugins/ChangeLogPlugin/ChangeLog.txt#L1-L9

I see in the history that both the entry for 3.0.3 and for 3.0.4 were added on the 17th of October, which was if I recall the release date for 3.0.4 (was it? Not 100% sure).

ChrisTerBeke, 2017-11-10:
3.0.4 release indeed has 3.0.3 and 3.0.4 listed in the change log.

closed 2017-11-10 by ChrisTerBeke