Ultimaker/Cura #2744: Setting "Initial Layer Z Offset" has no effect

TinkerGnome, 2017-11-06:
**Cura version:** 3.0.4
**Printer:** UM3
The setting "Build Plate Adhesion -> Initial Layer Z Offset" has no effect

**Steps to reproduce:**
- load an stl-model in Cura
- choose profile Fine - 0.1mm
- let it slice and save to file test_a.gcode
- change the setting for "Initial Layer Z Offset to 0.1 (e.g.), the value of "Z Offset Taper Layers" doesn't matter
- let it slice again and save to file test_b.gcode
- compare both files

**What results did you expect?**
There should be an offset of all z-coordinates visible in the second file (file_b.gcode)

**What is the actual result?**
Both gcodes are identical (except of the changed comments at the end of the file). The description of the settings says: "The extruder is offset from the normal height of the first layer by this amount.". This does not happen, the setting is completely ignored.

smartavionics, 2017-11-06:
Hmm, that's weird. You know I created those settings and some code that goes with them in the backend and I am sure that the backend mods were rejected by the Cura developers for some reason and not incorporated. Looks like they forgot to remove the settings from the UI. Sorry about that!

Ghostkeeper, 2017-11-11:
Huh, strange. The [related pull request](https://github.com/Ultimaker/Cura/pull/2176) wasn't merged as far as I could see.

Ghostkeeper, 2017-11-11:
I removed them here: https://github.com/Ultimaker/Cura/commit/6d05c320a83e77dc559db2bd2d32dc4beadaa28c

This is a nice bug report by the way.

closed 2017-11-11 by Ghostkeeper

Jason1712, 2017-11-12:
Hi,

Does this mean that cura will update to fix this issue, or do I have to apply this fix somehow? 

Cheers

fieldOfView, 2017-11-12:
It means that the settings will be removed from the next version of Cura, because it was decided not to add the functionality.

Jason1712, 2017-11-12:
Thanks mate

commit 6d05c320 (2017-11-11): Remove disfunctional initial layer z offset settings

The pull request for the engine wasn't merged. It looks like the pull request for the front-end settings wasn't merged either, but here they are. They shouldn't be here.

Fixes #2744.