Ultimaker/Cura #2745: Fedora 26, appimage 3.0.4

knuckleheadTech, 2017-11-07:
Cura from the repository works fine but I want to use 3.0.4 with the CR 10 settings. Running the appimage from a different computer (Xubuntu 14.04) works fine. But from on my Fedora 26 it starts to load then crashes as soon as the user agreement starts to come up. I never actually see the user agreement I just know from watching the other system run the appimage that it's the user agreement that crashes. Don't see any kind of warning, just kills off. 

Running from the terminal gives this output:

Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
Fontconfig warning: "/etc/fonts/conf.d/65-0-lohit-bengali.conf", line 32: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-sans.conf", line 24: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-serif.conf", line 24: unknown element "langset"
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
file:///tmp/.mount_Uzb2Us/usr/bin/resources/themes/cura-light/styles.qml:197:17: QML Item: Binding loop detected for property "width"
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
file:///tmp/.mount_Uzb2Us/usr/bin/plugins/plugins/UM3NetworkPrinting/DiscoverUM3Action.qml:282: TypeError: Cannot read property 'clusterSize' of null
./bin//cura.sh: line 11: 11489 Segmentation fault      (core dumped) cura $@

Appesteijn, 2017-11-08:
I made a pull request to fix this: https://github.com/Ultimaker/Cura/pull/2751

nallath, 2017-11-09:
This won't actually fix the crash. The typerror is just a warning.

mknj, 2017-11-14:
same crash for me. fedora 26 latest updates. 64 bit.

Ghostkeeper, 2017-11-22:
@nallath Does this happen on your work laptop?

It would help if someone would be able to run Cura under GDB and get us a stack trace.

nallath, 2017-11-23:
Nope. But I'm at fedora 25

etrigan63, 2017-11-24:
Same thing with Fedora 27 and Cura 3.04/3.1_BETA.

LipuFei, 2017-12-01:
@KurtLoeffler Could you try running it in this way?
`PYTHONPATH="" ./<your-cura-app-image>`
There was an issue with `PYTHONPATH` in AppImage.

LipuFei, 2017-12-01:
Managed to get a backtrace on a newly installed Fedora 27 x64. Looks like it failed on some Qt OpenGL module. I'll keep digging...
```
(gdb) bt
#0  0x00007f4569efc90b in raise () from /lib64/libpthread.so.0
#1  <signal handler called>
#2  0x0000000000000000 in ?? ()
#3  0x00007f4552836599 in QSGDefaultRenderContext::initialize(void*) () from /home/lfei/coredump/mount/usr/bin/libQt5Quick.so.5
#4  0x00007f455282adf7 in ?? () from /home/lfei/coredump/mount/usr/bin/libQt5Quick.so.5
#5  0x00007f455282b19d in ?? () from /home/lfei/coredump/mount/usr/bin/libQt5Quick.so.5
#6  0x00007f455ea9c98a in QWindow::event(QEvent*) () from /home/lfei/coredump/mount/usr/bin/libQt5Gui.so.5
#7  0x00007f4552898005 in QQuickWindow::event(QEvent*) () from /home/lfei/coredump/mount/usr/bin/libQt5Quick.so.5
#8  0x00007f455c86979c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /home/lfei/coredump/mount/usr/bin/libQt5Widgets.so.5
#9  0x00007f455c870710 in QApplication::notify(QObject*, QEvent*) () from /home/lfei/coredump/mount/usr/bin/libQt5Widgets.so.5
#10 0x00007f455cef4cfe in sipQApplication::notify(QObject*, QEvent*) () from /home/lfei/coredump/mount/usr/bin/lib/python3.5/PyQt5/QtWidgets.so
#11 0x00007f45600b8ab0 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /home/lfei/coredump/mount/usr/bin/libQt5Core.so.5
#12 0x00007f455ea923a9 in QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) () from /home/lfei/coredump/mount/usr/bin/libQt5Gui.so.5
#13 0x00007f455ea92eed in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
   from /home/lfei/coredump/mount/usr/bin/libQt5Gui.so.5
#14 0x00007f455ea70123 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/lfei/coredump/mount/usr/bin/libQt5Gui.so.5
#15 0x00007f4551fba6a0 in ?? () from /home/lfei/coredump/mount/usr/bin/libQt5XcbQpa.so.5
#16 0x00007f455f976bb7 in g_main_context_dispatch () from /lib64/libglib-2.0.so.0
#17 0x00007f455f976f60 in g_main_context_iterate.isra () from /lib64/libglib-2.0.so.0
#18 0x00007f455f976fec in g_main_context_iteration () from /lib64/libglib-2.0.so.0
#19 0x00007f4560106897 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/lfei/coredump/mount/usr/bin/libQt5Core.so.5
#20 0x00007f45600b6d9a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /home/lfei/coredump/mount/usr/bin/libQt5Core.so.5
#21 0x00007f45600bebc4 in QCoreApplication::exec() () from /home/lfei/coredump/mount/usr/bin/libQt5Core.so.5
#22 0x00007f455cef3f60 in meth_QApplication_exec_ () from /home/lfei/coredump/mount/usr/bin/lib/python3.5/PyQt5/QtWidgets.so
#23 0x0000000000572029 in PyCFunction_Call ()
#24 0x00000000004774c3 in PyEval_EvalFrameEx ()
#25 0x0000000000476bbc in PyEval_EvalFrameEx ()
#26 0x00000000004789cd in ?? ()
#27 0x0000000000478aa3 in PyEval_EvalCodeEx ()
#28 0x0000000000478acb in PyEval_EvalCode ()
#29 0x00000000004910bf in PyImport_ExecCodeModuleObject ()
#30 0x000000000050c6e4 in ?? ()
#31 0x0000000000572029 in PyCFunction_Call ()
#32 0x00000000004774c3 in PyEval_EvalFrameEx ()
#33 0x0000000000476bbc in PyEval_EvalFrameEx ()
#34 0x0000000000476bbc in PyEval_EvalFrameEx ()
#35 0x0000000000476bbc in PyEval_EvalFrameEx ()
#36 0x00000000004789cd in ?? ()
#37 0x0000000000478aa3 in PyEval_EvalCodeEx ()
#38 0x000000000054e385 in ?? ()
#39 0x000000000051d9ca in PyObject_Call ()
```

drindt, 2017-12-02:
Is there an expected Beta available soon?

Appesteijn, 2017-12-02:
You can download 3.1-BETA for the Ultimaker site since a week or so.

drindt, 2017-12-02:
But with the issue is still there.

knuckleheadTech, 2017-12-07:
Just grabbed 3.1.0 and still the same issue. Tried repeatedly trying to catch the user agreement by pressing Esc since in the release notes it says you can do that. But didn't make it that far. 

Thank you to everyone that has looked at this issues so far. 

Here is what I get from 3.1. Pretty close to my previous output.
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
Fontconfig warning: "/etc/fonts/conf.d/65-0-lohit-bengali.conf", line 32: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-sans.conf", line 24: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-serif.conf", line 24: unknown element "langset"
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
file:///tmp/.mount_VGecsP/usr/bin/resources/themes/cura-light/styles.qml:197:17: QML Item: Binding loop detected for property "width"
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
./bin//cura.sh: line 12:  7953 Segmentation fault      (core dumped) cura $@

LipuFei, 2017-12-08:
Hi, perhaps this PR https://github.com/Ultimaker/Cura/pull/2903 can fix this problem? If possible, could someone test this on your machine? You will need to run the code from source using the `master` branch.

ianpaschal, 2018-05-29:
This should be fixed as of the merging of that PR. If it's not, we can reopen.

closed 2018-05-29 by ianpaschal