Ultimaker/Cura #2749: Crash report

mennoberg, 2017-11-08:
Version: 3.0.0
Platform: Darwin-16.6.0-x86_64-i386-64bit
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/cura/PrinterOutputDevice.py", line 230, in setJobState
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/UM3NetworkPrinting/NetworkClusterPrinterOutputDevice.py", line 179, in _setJobState
    selected_printer_uuid = self._printers_dict[self._selected_printer["unique_name"]]["uuid"]
KeyError: ''

-------------------------
Thread 0x000070000928a000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008d87000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008884000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008381000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700007e7e000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000078f8000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000073f5000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700006ef2000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

[... 33 lines cut ...]

Thread 0x000070000505d000 (most recent call first):
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007fffd3e213c0 (most recent call first):
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 100 in show
  File "/Users/ultimaker/build/3.0/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 686 in run
  File "/Users/ultimaker/build/3.0/build/inst/bin/cura_app.py", line 82 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

ChrisTerBeke, 2017-11-08:
I see this is version 3.0.0. Can you try 3.0.4, we did some bug fixes in there that solve an issue that created reports similar to this one.

Ghostkeeper, 2017-11-14:
What were you doing when this crash occurred? How should we reproduce this?

ianpaschal, 2018-02-28:
Needs more info but never got it.

closed 2018-02-28 by ianpaschal