Ultimaker/Cura #2754: Mac 3.0.4 version weirdness

kentonbmax, 2017-11-09:
- Windows seems to print better with same settings, same printer, same filament.
- Start up sequence on windows raises then goes to middle, mac goes out 50 on x then 50 y instead of a perimeter around the part. 
- `Fine` is showing .2 and has .2mm settings on mac. Should be .1mm like on windows?

kentonbmax, 2017-11-09:
[image]

LipuFei, 2017-11-09:
@kentonbmax You are right. In the screenshot, the selected `0.2` should actually be `0.1`.

LipuFei, 2017-11-09:
Hi @kentonbmax, I confirm that this bug is in 3.0.4 (also on Windows) but is fixed in the master branch by this commit e897431c959491a8cf27fbf275bd2bcabdb15d01.

This problem will be solved in the next release.

kentonbmax, 2017-11-09:
Great! I'm really a fan of the startup sequence and the fine settings on windows.

Ghostkeeper, 2017-11-15:
That fix only fixed the layer height, not the start-up sequence though.

If you have a different starting sequence on different platforms, I think the start-up sequence might have accidentally changed on your mac. Try this:
* On both computers, click on the printer name (Creality CR-10), and select "Manage Printers..."
* Click the Machine Settings button.
* Compare the Start Gcode field in the dialogue that appears. If it's different, then you might want to take your version on Windows.

jackha, 2018-07-12:
Is this still a thing? I would like to close the issue.

jackha, 2018-07-12:
No activity and we're now on 3.4.1. Is it still something? I would like to close it.

kentonbmax, 2018-07-12:
Will try this weekend.

nallath, 2019-01-07:
Closing due to inactivity.

closed 2019-01-07 by nallath