Ultimaker/Cura #2755: Layer thickness order in Cura 3.04 is wrong

MyGitCat, 2017-11-09:
Hi,
there is a wrong order in Layer thickness (Schichtdicke): 0.06 0.25 0.15 0.2 0.4 0.6.
My be that only the text in the window is wrong ...
[image]

fieldOfView, 2017-11-09:
What printer definition do you use?

This happens when a printer definition (the .def.json file) contains a default layer height. Instead it should specify a default quality. See this commit that fixes it for the Creality CR-10 definition:
https://github.com/Ultimaker/Cura/commit/e897431c959491a8cf27fbf275bd2bcabdb15d01

LipuFei, 2017-11-09:
@MyGitCat I guess you are using a Rigid 3D? The whole family have default layer_height = 0.25.

MyGitCat, 2017-11-09:
I am using a 3DP-100. Default layer-height sounds very good, but I coud not find it in cura 3.04 which is very new for me. I will search for it and I am sure that is my problem.
Thank you so far.

MyGitCat, 2017-11-09:
Hello fieldOfViwe,
thank you for your info. I did not know that there are a lot of JSON printer files inside cura. I think I have to create my own 3DP-100.json now. Is there something like a default.json printer? And if where is it defined?

fieldOfView, 2017-11-09:
>  I think I have to create my own 3DP-100.json now.

Not necessarily. You have not answered the question what *definition* you used. IE: when adding your printer to Cura, what printer type did you choose from the list?

MyGitCat, 2017-11-09:
My Printer was not part of the list. I added my own printer configuration "Daycom" with 15.04.6 or earlier. And "Daycom" was the automatic default in Cura 3. Maybe it is better to configure a new printer in Cura 3.

MyGitCat, 2017-11-09:
The problem is fixed!
I had to configure a new printer (Custom FDM Printer). I used the same values as before. After this session the order was correct! 
By the way: I thought that the thickness values comes from Cura 3 but Cura used the values I defined 2 years ago with an old cura version (protected profiles).

--- Thank you ---

closed 2017-11-09 by MyGitCat