Ultimaker/Cura #2756: Cura 3.04 print starts in the air

MyGitCat, 2017-11-09:
Hi,
I am using a DAYCOM 3DP-100 and changed from Cura 15.04.6 to Cura 3.04.
Printing with old gcode works fine but new gcode fails because the print head is a little bit to high.
I checked the difference between both gcodes. The third last line (attachment) shows what is going wrong.
Old gcode: G0 ... Z0.300  (print is OK)
New gcode: G0 ... Z0.5    (prints in the air)
I am sure that this must be a configuration problem but I have no idea how I can fix it.
Regards,
Willi

[image]

LipuFei, 2017-11-09:
@MyGitCat Could you check the setting "Initial Layer Height"? The number there could be wrong.

MyGitCat, 2017-11-09:
I am sure that "Initial Layer Height" is my problem. In older versions this setting was on the “Advanced” tab, in the “Quality” section but where is it in Cura 3 ?

fieldOfView, 2017-11-09:
In the "Custom" tab, there's a search box.

MyGitCat, 2017-11-09:
Thank you - I ignored the custom tab - why ever.
After changing "initial Layer Height" to a smaller value I saw this small value (and not 0.5) in gcode. But without success - the printer prints in the air again.
What I did now: I changed the temperature from 210°C to 220°C and slowered the speed to 30 mm/s. Now it works. Now I have to check all my settings in the custom box.
--- Thank you ---

closed 2017-11-09 by MyGitCat