Ultimaker/Cura #2760: Mac 3.0.4 Monitor Page Incorrect Target Temperature + Missing Job Name

EnumC, 2017-11-09:
Hello everyone. I noticed an issue with Cura 3.0.4 on Mac, and as you can see in the screenshot below, the target temperature display "0C" for both extruder and build plate despite it properly being set, indicating a UI issue? Job name, print name, estimated time left are also either empty or at 00:00:00. Is this the normal behavior for anything that's not an "official" ultimaker?

[image]
)

Vandrasc, 2017-11-10:
Hi @EnumC ,

If connected to a printer via USB, the target temperature shows a temperature of 0°C. This is an unfortunate by-product of how the USB connection was implemented and the fact that there is no g-code to obtain this data. At least this was the idea behind it when we implemented this feature. If any thoughts/ideas please feel free to share :)

The second issue, show time estimate and print job name in wifi printing and USB printing, it is a known issue, our product manager just removed this from our planning for being inactive for too long. https://github.com/Ultimaker/Cura/issues/1448

EnumC, 2017-11-10:
@Vandrasc Doesn't Marlin regularly return current temperature and set temperature in ACK messages? Wouldn't there be a way to grep a certain part of that message which contains the set temperature from serial?

fieldOfView, 2017-11-10:
Marlin only reports the current temperature, not the target temperature

EnumC, 2017-11-10:
I thought it returns something like this:

`T:23.61 /0 @:0 T0:23.61 /0 @0:0 T1:23.89 /0 @1:0`

Where `/0` is the target temperature?

ChrisTerBeke, 2017-11-11:
That depends heavily on the Marlin version and manufacturer modifications. I've seen so many different responses over the years, including T:XX / X, T:XX, TX, T0X, etc. To find your printer's version try hooking it up to Repetier Host and check the serial console.

fieldOfView, 2017-11-12:
@EnumC, you are quite right, at least in the case of the UM2 and UMO. I will see what I can do.

PR #2773 (2017-12-01): Show USB target temperatures, (estimated) time and jobname
This PR enhances the USB Printing plugin to show target temperatures (from M105 responses), shows the total and estimated remaining time, and the jobname while printing. This was previously left unimplemented.

This PR fixes #2760
closed 2017-12-01 by ChrisTerBeke