Ultimaker/Cura #2761: Suggestion: "Sloped" Z-Hop

Thorinair, 2017-11-10:
I've been using Cura for a while now and I really managed to beautifully tune my prints, but one thing keeps giving me trouble: Z-Hop. One big drawback of Z-Hop is that it introduces lots of stringing when the nozzle is lifted. Theoretically, this is mostly because of that initial lift. Now, I haven't managed to find anything like this on the net or in other slicers, but I propose a solution that I would call "Sloped" Z-Hop.

[image]

The idea is to completely avoid the starting lift and have the nozzle move in a slope to the next part before being lowered. The drawback of such an option would be that there might be a bit of nozzle drag on the initial move, but the upside would be that it should theoretically eliminate stringing.

Thoughts?

DDDirk, 2017-11-11:
Sounds like a good idea, although in your version the angle of the slope depends on the distance. Maybe with some experimenting an ideal angle can be found (probably depending on other factors), which can then be used for an initial move, followed by a horizontal one towards the target. (And if that is too close, the initial move may be in the opposite direction.)

Thorinair, 2017-11-11:
Indeed, it would be rather steep when printing something like supports. Extra settings would definitely be needed. A horizontal move and then z-hopping regularly would also be an interesting solution. It would add a bit to the print time, but possibly help with stringing a lot.

Ghostkeeper, 2017-11-17:
We do a sort of horizontal move at the end of a print to prevent the little string at the top of the print. We should probably do something similar for this. But does it give a lot of problems then? Usually the z-hop should occur inside the model, right?

Ghostkeeper, 2018-02-05:
Our project manager just killed this feature request because it's almost older than 12 weeks.

jflyer81, 2018-12-26:
It seems that this feature request has been deferred but I would love to see a "wipe while retracting" feature as well where Cura starts the horizontal retraction move and the z-hop vertical move simultaneously.

RedPetroleum, 2019-01-27:
I just signed up for Gitup just to say that I really like the idea of sloped z hop.
I hope this will be made because I got some problems with stringing caused by z hop that I cannot solve. 
Also wipe while retract would be awesome!

Ghostkeeper, 2019-01-29:
If you have some print results to share, we'd love to see them!

yeowjoon99, 2019-11-16:
@Ghostkeeper I think the reason Z hop in Cura causes so much stringing is that when Z hop is enabled, it doesn't actually do the wipe at the end of an 'island'

Not sure if anyone already mentioned that but yea it's something I just noticed after looking into Z hop and stringing

Goury, 2021-12-20:
Bump.
Why is this not yet implemented?

Ghostkeeper, 2021-12-30:
Because someone would need to implement it.

This issue is deferred, meaning that we do see some value in it, and would like to get it implemented eventually, but it currently doesn't have priority. We get like 4x as many feature requests as we can implement, and additional feature requests from Ultimaker itself, so we have to choose.