Ultimaker/Cura #2762: Per Model Setting default unchecked - Generate Support

nicolinux, 2017-11-10:
There is a problem with "per model settings" where the default unchecked state of a settings does not produce the desired outcome. For example when two objects are added and only one should not be printed with support. If "generate support" is enabled globally and then per object setting for one object is set to "generate support" - it is unchecked in the default state. Cura would still slice with support. One needs to check once the per model setting and un-check it again for the UI to pick it up.
For a real-life case, please see this forum post:
https://ultimaker.com/en/community/51898-about-the-supporting-material

fieldOfView, 2017-11-10:
The same is true for other setting, eg if the infill density is set to 30% in the sidebar, and I add the infill density to the per object settings, I expect it to show 30% initially. Instead it shows 20%

fieldOfView, 2017-11-10:
This works correctly for single extruder printers

Ghostkeeper, 2018-02-05:
This doesn't work correctly any more for single extrusion printers since that was streamlined in 3.0 to behave the same.

Ghostkeeper, 2018-02-05:
This bug was removed from our planning since it was untouched for 12 weeks.

fieldOfView, 2018-02-05:
“Ultimaker Cura, now equally broken for multi- and single-extruder printers”

Ghostkeeper, 2020-08-14:
This is fixed in more recent versions.

closed 2020-08-14 by Ghostkeeper