Ultimaker/Cura #2763: [3.0] Tweak UI to fit 'Custom Setup' section better on small screens

lucaspiller, 2017-11-11:
I'm running Cura on a 13" Macbook Air with a 1440x900 screen. Other Macs have even smaller vertical height. The changes in 3.0 have expanded some elements while shrinking the 'Custom Setup' section a lot - which is the main thing I use on the sidebar. Here a side by side view of 2.7 vs 3.0 with the window maximised:

[image]

Is it possible to optimise the sidebar, so this section better fits on small screens?

fieldOfView, 2017-11-11:
See #2678

ianpaschal, 2018-05-26:
Sidebar compactor is now a thing and we won't be redesigning the UI much prior to 4.0 at least. Also this is super old.

closed 2018-05-26 by ianpaschal