Ultimaker/Cura #2764: Move tool minus sign not visualized

BagelOrb, 2017-11-11:
[image]

Y should read -16, not 16

Linux Mint
latest master
Lenovo P51s

smartavionics, 2017-11-11:
That's interesting, it seems that the sign of Y is inverted?

[image]

smartavionics, 2017-11-11:
Here's another picture with the camera pointing along the Y axis. In both this and my previous picture compare the direction of the green arrow with the green axis marker!

[image]

BagelOrb, 2017-11-11:
I don't think that's related to my issue.

I dont think what you are describing is really a problem.

smartavionics, 2017-11-11:
I agree it's not a problem inasmuch that it doesn't affect the print but why do the Y axis arrow and marker point in opposite directions when the X and Z arrows and markers have the same sense?

Back to your issue, does the - sign never appear, irrespective of where the part is positioned relative to the axis markers?

Ghostkeeper, 2017-11-17:
There are two major coordinate systems in Cura. One that's used by the front-end which is intended to be the same for all printers. The other is used by CuraEngine and is the same as the printer's coordinate system.

As for the actual issue, it doesn't reproduce on my Ubuntu 17.10 so it's either something with Mint or it's something with your graphical pipeline.

ianpaschal, 2018-03-09:
@BagelOrb still having this issue or can I close? I'm not able to reproduce it either.

BagelOrb, 2018-03-09:
Can't reproduce either.

closed 2018-03-09 by BagelOrb