Ultimaker/Cura #2765: 3.0.4 Stops in the middle of the printing

ZerialLim, 2017-11-11:
Its very strange, but the printing process stops in the middle of 10+ hours printing, making just a rubbish.

using different setting values, stops again.
using different STL, sometimes fine, but I need to print this one...
using same GCode after cleaning up the hardware, >STOPS AT THE SAME POINT<

as cura does not support any terminal function I tried the same GCode to use it on MatterControl to monitor the console message and...
...It's just fine.

What should I provide to prove this error?

ChrisTerBeke, 2017-11-11:
You're using the print over USB feature in Cura?

ZerialLim, 2017-11-11:
@ChrisTerBeke // Yes because my printer does not support SD Card.

ChrisTerBeke, 2017-11-11:
USB printing is fairly unreliable. Cura also doesn't support sending the G-code lines safely (using N-codes), so it can potentially fail at any time. Repetier host does support safe sending, not sure about MatterControl. Since Cura is primarily a slicing application the USB printing was mainly a community contribution and I don't expect it to further improve anytime soon. For now I suggest you use Cura to slice and send the G-code using MatterControl or another host software.

closed 2018-02-26 by ianpaschal

Ghostkeeper, 2018-03-16:
If you're using a smoothieware-based printer, we've just merged in some improvements here to fix an issue that had the same symptoms as you: https://github.com/Ultimaker/Cura/pull/3346
That should get available with the next release.