Ultimaker/Cura #2766: SPiralize Outer Contour - Single wall limitation?

Thisismydigitalself, 2017-11-11:
I am printing an hollow object (Not a vase) which could be printed in "Vase" mode but when i chose "SPiralize Outer Contour" it seems to reduce wall count to 1. this sounds reasonable but what if i want more than a single wall? i am trying to get around hundreds of travels, each taking place at the same place - mid circumference which by then the nozzle travels across to the other side to resume printing - i don't find the logic in that.

Is there a setting to overcome this?
Thx.

smartavionics, 2017-11-11:
The short answer is, no. Spiralize can only produce a single wall. What I have done before is to make the wall line width wider (i.e. use 0.8mm rather than my normal 0.5mm).

Your comment about hundreds of travels is interesting. If you are using Cura 3 and have enabled the wall order optimization you should not be getting lots of travels between inner and outer walls on a hollow object. Please check your settings and if you are still having a problem then please make the model available or at least a screenshot of the layers. Thanks.

Thisismydigitalself, 2017-11-11:
Bingo. you are correct. i did not know about this Optimization thingy.
here it is with and without. big differences tho i still wonder why the nozzle needs to jump when it looks like it can continue its path in a long spiral. anyways, i will give this a try in my next print.
Thank you.
[image]
[image]

Thisismydigitalself, 2017-11-11:
In addition, i just tryed your second tip - Line width.
Works nicely (still don't know how the outcome will look like) but what i've noticed is that the Estimated print time is not reflected as it should when increasing line width in Spiralize mode.

smartavionics, 2017-11-11:
Looking at your pictures, that model is not suitable for being spiralized because at the bottom it has multiple parts (the dangly bits that hang down). By the very nature of how it works, spiralize is only going to work OK when each layer contains just a single part and the model does not have surfaces that slope less than about 45 degrees (vertical surfaces are good, horizontal are bad).

closed 2017-11-14 by ChrisTerBeke

Ghostkeeper, 2017-11-22:
In my experience, having multiple parts with spiralise sort of works but then you'll have a seam so you're off just as well as disabling spiralise.

I suspect that the print time is reduced by increased line widths because it reduces the number of skin lines it has to draw. If you're only adjusting the wall line width it could still have that effect by reducing the skin areas themselves.

agasevangelos, 2018-03-22:
Hi I have the same problem!
I need a model of a detergent bottle with a thickness of 1.2mm to be printed with no infill. 
Cura makes a crappy toolpath and the results are horrible.
Slic3r manages to produce a beautiful continuous toolpath and as a result and very good part.
Is there any solution to overcome this bug on Cura?
[image]
[image]
[image]

smartavionics, 2018-03-22:
BTW, I have created a PR (https://github.com/Ultimaker/CuraEngine/pull/721) that helps it cope better with an odd number of wall lines but it hasn't yet been approved/used.