Ultimaker/Cura #2767: Improvement Request: Material Type <-> Profile Correlation

Schwebewerk, 2017-11-11:
Dear Community,

is it correct, that profiles are linked to material types? I am not aware of any indicator within the software, but changing a profile seems to apply only for same material types. If profiles and mat types are linked like this, then:

-> How can I create a new Material Type? I want to bring Polycarbonat for example into Cura, but Profiles are unavailable unless I pretend my new Material to be something existing like ABS or PLA.

-> Could you improve Cura by adding a headline in the profile editor, which indicates the material type, which is being affected?

I am using Cura 2.7 and the recently released BCN3D-Sigma Cura Version.

Best, Stefan

Schwebewerk, 2017-11-11:
Follow up Question including the correlation between profiles and multiple extruders [#2048](https://github.com/Ultimaker/Cura/issues/2048):

-> If profiles are linked to material types && profiles are always containing all extruders, then what exactly happens, when i put different material types on different extruders? Does this creates a profile which relates to two separate material types?

ChrisTerBeke, 2017-11-12:
* You can enter the material type when creating a custom material. As this is a text field you can enter any type you want.
* The profiles you see when editing are the profiles that match the current selection of materials. When creating a new profile, it always is targeted at the material currently selected. We're probably changing this in the future to show materials and profiles regardless of current settings in the preferences window and indicate their relationship.
* Profiles are derived per extruder (for the selected print core + material), and a common profile is found (global profiles) that match both by quality type. We're looking at several ways to improve this in the future, like profiles per material combination, but nothing concrete yet.

closed 2018-03-02 by ChrisTerBeke