Ultimaker/Cura #2768: CURA Infill overlap and skin overlap percentage is not present in the printing

Danielrug05, 2017-11-11:
Hi, 

I am trying to print with CURA 3.0.4 and this happened:
[image]
[image]
[image]

In one of them you will see the same piece printed with CURA version 15.04.2 and the other with 3.0.4. When i tried to print FLEX, using the 3.0.4, the skin, inner wall and infill are separated from each other as you can see. The printed piece has numerous flaws compared to the one printed with the older software. what can i do to correct this? i have tried increasing the infill and skin overlap percentage up to 20% and the result is the same.

Thank you.

daveposh, 2017-11-12:
I'm seeing the same in my prints. Going to switch away from the concentric shell pattern and see if it helps.

Danielrug05, 2017-11-14:
Well please tell me if yousee any difference. I already tried with PLA, to see if the one to blame was the material but the same happened. Any suggestions?

Chompworks, 2017-11-16:
Also getting similar issues with both concentric and lines infills on top skin.
First noticed it with Ironing enabled, then assumed it was a bug in Ironing, so disabled ironing and hit the same issues.

On my prints it looks like a poorly extruded first wall that connects to the infill, but not entirely sure that thats accurate, as similar retraction settings from older, working Cura versions don't cause the same issue. 

As you've said, infill overlap settings don't seem to affect/improve anything.

nallath, 2017-11-16:
I've not noticed any of these issues, but I'm running stuff directly from master. Does anyone have issues with older vesions? I would help if we could pinpoint where this issue occurs. Also; Does this issue occur on only non-um printers (as that would explain how we never found it, as we only test on those)

Chompworks, 2017-11-16:
Must admit, didn't see it in v3.0.1 or 3.0.2 - skipped 3.0.3 direct to 3.0.4. 
Have also deleted all prefs on mac, reverting cura to a clean install state, problem still occurs, just in case it was that skipped point release.

Will try on our UM2+ hopefully later today, just finishing some customer orders on it first.

Chompworks, 2017-11-16:
Just tried a small quick print on our UM2+ - can't speak for the others, but here, it doesn't seem to have the issue, so it might be only on non-ultimaker machines?
Will try a different model with a few more interesting curved walls [the test I used was basically cuboid] later on today.

nallath, 2017-11-16:
If there are differences, they will be between 2.7 and 3.0.x. The difference between minor version changes is usually quite small.

Danielrug05, 2017-11-16:
Well, i dont have an ultimaker and probably is because of that. So this issue only dissapears when you use a UM2+? its a shame because i have to do some precision work and, though i can work with the 15.04.2, the features that i can control like the roof in the support material could improve a lot the quality my printings.

Ghostkeeper, 2017-11-25:
If you're using concentric skin, the skin overlap setting will disappear. They are not supported in combination at the moment.

With the lines skin pattern, the skin overlap seems to work fine. Here's with 100% overlap:
[image]

And with the default 5% overlap:
[image]

I think that you're either trying to edit the infill overlap to modify the skin overlap, or that your line width is different from what it was in Cura 15.04 and that's what causes this issue.

Danielrug05, 2017-11-30:
Hi, 

When you say concentric skin, you mean that the printing have a concentric form? i looked for patterns in the configuration settings and the only ones that appeareand, that are of my concern,  are top skin surface pattern and infill pattern, and i have both of them with lines.

Ghostkeeper, 2017-12-13:
> When you say concentric skin, you mean that the printing have a concentric form?

Yes. The lines are basically like walls going all the way in.

As far as I could see the skin overlap is working perfectly in the g-code output. Maybe you can provide a project file where you enabled skin overlap but it's not showing in the output?

Danielrug05, 2017-12-14:
Hi, 

i will send a very simple file, the one i used when i detected the error.

please let me know what you find.

[test file.zip](https://github.com/Ultimaker/Cura/files/1560433/test.file.zip)

Ghostkeeper, 2018-01-02:
My findings are similar. Skin overlap seems to work properly. Here's 100% overlap.

[image]

I think it's just a matter of increasing the overlap on your print. Can you indicate where the g-code is wrong? Is Cura doing anything unexpected?

Danielrug05, 2018-01-03:
well, i suppose the gcode wasnt wrong, the issue was the skin wasnt overlapping once i printed the piece. Did you try to print it?

Ghostkeeper, 2018-01-22:
It looks all right.
[image]

closed 2018-03-26 by ianpaschal

evgen48, 2018-11-17:
> If you're using concentric skin, the skin overlap setting will disappear. They are not supported in combination at the moment.

Hello, @Ghostkeeper do you know will this be fixed near soon? Cura 3.6.0. still doesn't have a fix. Currently I use 100% infill, tried also 110%, and increased infill line width, about 3x compared to nozzle diameter of 0.4, while printing 10mm cylinders, so I only can use concentric infill to minimize gaps,  however lack of infill  overlap setting gives me a big gap in center. Thank, you.