Ultimaker/Cura #2770: Improvment request

Thisismydigitalself, 2017-11-12:
I happens more and more that i need to split STL's in half in order to print those in sections.
what i do is lower 50% down through the print bed and then rotate 180 to print the other side.
this works great for rounded objects but when the need comes to split cubical objects it wont work cause i need to flip those in a way the open part is at the top and the face is touching the bed. 
in other words - if i could only lower the objects through the bed, then rotate it but this time Cura will ignore the part that was under the bed - like it was hidden or cut out - that would be great.

ChrisTerBeke, 2017-11-12:
Lowering objects below the bed is already supported. You might have to turn of the 'drop models on bed' feature to make it work. Everything below the print bed will be ignored by the engine.

Thisismydigitalself, 2017-11-12:
> what i do is lower 50% down through the print bed and then rotate 180 to print the other side.
> 

> Lowering objects below the bed is already supported.

> 

If that was not clear enough, this is what i practice these days.
If you'll read again you will see that what I'm looking for is a way to not only drop through the print bed but also to be able to rotate 180 degree the remaining part of the model which is above the bed in such a way that the part under the bed will not be visible nor printed. 

basically saying: turn the bed to split objects.

ChrisTerBeke, 2017-11-13:
ok, sounds like a way to have model cutting, but in an interesting way. I know we did some research on cutting meshes before, but it never made it's way into a full feature. Looking at the new clustering functionality of Cura Connect, I'm sure we'll get back to it sooner or later.

fieldOfView, 2017-11-13:
Didn't @nallath work on cutting objects?

nallath, 2017-11-13:
Yeah. But it was rather crash prone. Never got to actually fixing it. I made the stupid mistake to use cgal, which is horribly documented & unstable as can be. The library is clearly not meant for production code of any kind (It's used quite a lot by academia, where stability isn't really a thing).

fieldOfView, 2017-11-13:
> It's used quite a lot by academia, where stability isn't really a thing

\*shots fired\*

nickthetait, 2017-11-13:
duplicate of https://github.com/Ultimaker/Cura/issues/2472

Thisismydigitalself, 2017-11-13:
Bottom line: 
If Cura supports cutting objects then - How?
If not - Case can be closed but i would love having this option in Cura.

fieldOfView, 2017-11-13:
There is a way, but it is not pretty.

* Load your object, and a cube/box into Cura. 
* Scale the cube/box so that it can overlap the part you want to cut away
* Group the object and the cube/box
* Ctrl-click the cube/box (so it is selected, while grouped), and move it over the part you don't want to print
* Open the Per Object Settings tool (with the cube/box still selected), and add the following settings:
Cutting Mesh, Infill Density, Wall Thickness & Top/Bottom Thickness
* Set Cutting Mesh to checked, and the other 3 settings to 0
* You have now effectively subtracted the cube/box from the object.

ChrisTerBeke, 2017-11-13:
Closing this issue as it's a duplicate. Any further discussion can still be held here if wanted.

closed 2017-11-13 by ChrisTerBeke

Thisismydigitalself, 2017-11-14:
Thank you. i understand the logic and will give it a try but it takes less time to cut such using MeshMixer. i was hoping Cura can do this as it already ignores what's under the bed - it's now just a matter of flip the remaining part above the bed 180 facing up and - we're there.