Ultimaker/Cura #2771: GUI lag

Dhdrummer, 2017-11-12:
If there's model on the build plate or not, GUI of 3.0.4 is very, very laggy. Every time I tried to change a setting it freezes for 1-3 seconds. And that's REALLY ANNOYING if I want to change lots of settings. I'm running Win10, i5 with 8GB of RAM, SSD and HD6850M on my laptop. I hope the reason for this lag is my outdated graphic card.

ChrisTerBeke, 2017-11-12:
Try turning of auto-slicing, that might prevent some issue on lower spec machines.

Ghostkeeper, 2017-11-20:
We've noticed that the rendering of layer view was still continuing even if the user is not in layer view. That should be fixed in version 3.1.

dmitrygorbenko, 2017-12-26:
I'm using version 3.1.0 with model view active and no models on a table at all -- so, nothing to slice for Cura (just launched it), and if I change any parameter like speed or temperature Cura freezes for 4-5 seconds.
It's so annoying.
But I keep using it because it's much better than other software I used.
Please, fix these lags asap !

DanalEstes, 2017-12-27:
I have this same symptom.  Very, very, very laggy UI.  Several seconds PER DIGIT in any input field. Upwards of 10 to 20 seconds to change certain things, such as a material.  Auto-slice is turned off. 

All other 3D software runs fine on this hardware.  Fusion 360, OpenSCAD, Sketchup, Slic3r, you name it, all fine.  Heck, all other software runs fine.  Also, Cura slicing seems to run in reasonable times, of course varying somewhat per model.  Any given model, when compared to other slicers, seems about the same in Cura.  The "orbit" and "slew" controls when changing views of the bed within Cura are also good, smooth, and responsive. So I don't think it is the hardware. 

Windows 10, Cura 3.1.0.  Any 3.x has been very sluggish, about the same across releases. 

Many things I like about Cura, including the spirit of openness.  However, the almost unusable UI is driving me batty. 

I'd be happy to post any logs, etc.

Ghostkeeper, 2018-01-16:
Maybe a project file could allow us to reproduce it then. Could you open Cura (with an empty build plate) and click File -> Save Project, then upload that here?

fieldOfView, 2018-01-16:
Some more steps, for yet more chances of finding out what is happening:
* Open Cura (with an empty build plate)
* Change a setting
* Click File -> Save Project
* Close Cura

In addition to the project file, please include the stderr.log and stdout.log files from %APPDATALOCAL%\cura in a zip and attach here. (Note: that is c:\users\[you]\appdata\LOCAL\cura, not roaming; the local folder contains the logs for only the last session)

DanalEstes, 2018-01-16:
I think you actually asked for these files from someone else... just in
case, here is another data set:

Computer is an Intel I3, 8 Gb Ram, Intel built-in graphics, Windows 10
Home, recent patches/upgrades.  This machine runs other 3D Print software,
such as Fusion 360 or OpenScad or Slic3r just fine.  Cura slice times are
as expected; GUI is slow to very slow.  I also run Cura on a Macbook pro,
and it runs fine there.

For these logs:

Opened Cura, no model loaded.  Changed Layer Height.  About 3-5 seconds of
"clocking" mouse cursor after change.  Then changed material temperature.
About 3 seconds per character typed, and then when mouse clicked in the
next field, about 5 seconds before cursor moved.

Project and logs attached. (by the way, it is  %APPDATA%\..\local\cura).

Thanks for continuing to pursue this.

Danal

P.S.  You may see in the logs both the "Octoprint" plugin, and I'm working
on a similar plugin for "Duet3d".  The performance problem exists with or
without both of those.  I'd be happy to send logs with them removed, if you
prefer.

fieldOfView, 2018-01-16:
Unfortunately, sending attachments through email does not work on Github. Please post them through the github.com web interface, or upload them somewhere else and post a link here.

DanalEstes, 2018-01-16:
Will do!

DanalEstes, 2018-01-16:
Here are the files for the email that's about two or three above. 

[TestCura.curaproject.3mf.txt](https://github.com/Ultimaker/Cura/files/1635709/TestCura.curaproject.3mf.txt)

[stdout.log](https://github.com/Ultimaker/Cura/files/1635693/stdout.log)

[stderr.log](https://github.com/Ultimaker/Cura/files/1635692/stderr.log)

DanalEstes, 2018-01-16:
I un-installed Cura (windows installer) and then used file manager to search "user" and "program files" and so forth and remove ALL traces of Cura. 

Then downloaded and installed, taking all defaults, except un-checked "install Arduino Drivers". 

Opened Cura, Defined one custom printer.  Defining the printer, each field taking several seconds.  Saved printer, went to "custom" and changed layer height.  Several seconds delay.  Changed material temperature, noticeably more delay to change a 3 character field, and several seconds to tab to next field. 

Log Files attached.  (forgot to save project file, hopefully not important because of clean install)

[stderr.log](https://github.com/Ultimaker/Cura/files/1635836/stderr.log)

[stdout.log](https://github.com/Ultimaker/Cura/files/1635839/stdout.log)

Let me know if you need anything else.

nallath, 2018-01-17:
Ugh. I don't see anything in the logs that could explain the issue.

jackha, 2018-01-22:
@DanalEstes Thanks for the explanation and the project file. It seems that the problem was that it was error checking all extruders defined in Custom FDM printer, 8 of them. In the upcoming 3.2 beta this is fixed.

For the other people: what printers are you using? Is it also Custom FDM Printer? Then it should be fixed. If it's something else, I'd like to see a project file so I can debug it. Thanks!

jackha, 2018-01-22:
devs: it's CURA-4785

DanalEstes, 2018-01-30:
> Just tried Cura-3.2 beta on Ubuntu 17.10 and it works much faster !!
> A lag is about one second which is almost OK, but can be better :-)

3.2 Beta is better on Windows 10 as well.  A little better.

I use a custom printer definition, and a different github issue thread
indicated that:

* All custom printers have 8 extruders.
* There are numerous error messages in the logs regarding extruder 2
through 8.
* The non-existant extruders seem to cause delays as parameter changes are
validated.

3.2 beta is better, a little.  Deleting extruder 2 through 7 in all the
configuration files makes 3.1 or 3.2 Beta MUCH MUCH better.  Either one is
entirely usable, after deleting these extruder definitions.

Thanks,

Danal

jackha, 2018-01-31:
@DanalEstes Deleting extruder 2 - 7 making it much much faster is very interesting! That indicates that Cura still does all kinds of things with them. I suspect that it has something to do with signals being handled, which may be a bit risky to change in the beta.

closed 2018-02-28 by ianpaschal

bennett12349, 2019-03-21:
new to cura and just downloaded on my macbook pro with a solid state Hard drive and 16GB of ram. The program as a whole is extremely slow. I made sure auto slicing was turned off. I downloaded the program on my kids super cheap Windows laptop and its much fast with a lot worse specs. None of my other programs run that slow for me. Any ideas?

fieldOfView, 2019-03-22:
There's an OSX-specific issue regarding the rendering of the 3d scene. More info here: https://github.com/Ultimaker/Cura/issues/4505#issuecomment-451858287

bennett12349, 2019-03-23:
Alright I'll look into that. Thanks.