Ultimaker/Cura #2772: gcode gen by CURA3.0.4 cause Marlin to reset during print.

dream4magi, 2017-11-12:
[CFP_The_Ultimate_Spool_Holder_v2.zip](https://github.com/Ultimaker/Cura/files/1465021/CFP_The_Ultimate_Spool_Holder_v2.zip)
[image]

Marlin resets at the same line every time.
I don't know if this is a bug of Cura or Marlin.
This line seems to be a normal G1.

ChrisTerBeke, 2017-11-12:
Looking at the G-code, which is completely normal, I'd say it's a firmware bug. How about the hidden characters (like new lines or end lines)? An editor like Notepad++ can show those for you. It should have a `\n` at the end of every line.

smartavionics, 2017-11-13:
The gcode does look OK in craftware gcode viewer. Have you tried moving the print on the build plate or changing any print settings to alter the speeds/line widths, etc?

dream4magi, 2017-11-13:
After then, i rotate the model and slice again.
The printer didn't reset.
I do think the bug is belong to Marlin, since the gcode is so normal.

closed 2017-11-13 by dream4magi

Ghostkeeper, 2017-11-20:
You might want to report this in Marlin's repository?